OpenIntelWireless/itlwm #971: WIFI6(E)连接不上

xjh169, 2024-03-21:
说明型号
为配备第 8 代 i7 的小米-Pro MX250

说明 EFI 的版本
XiaoMi_Pro-KBL-OC-v1.8.2.zip
网卡：英特尔® 双频带 Wireless-AC 8265
测试的系统版本：macOS14.4、macOS13.6.513.6.5 (22G621)

Wi-Fi 在5GHz频段 目前为止测试https://github.com/OpenIntelWireless/itlwm/releases，没有连接成功
[image]
现在只能在ac模式下才能使用5GHz频段 AX模式就无法连接了
[image]

zxystd, 2024-03-21:
抓日志：https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

xjh169, 2024-03-21:
在观察中，ax模式下 有时不兼容ac。。

zxystd, 2024-03-21:
@xjh169 不是，目测是你的路由器的问题，你选的那个模式仅支持11n、11ac、11ax但是不支持11a，itlwm是先关脸上11a然后再升级到11n/ac/ax，没有11a的话itlwm是连不上的，跟11ax没有关系。

xjh169, 2024-03-21:
> 好，知道了。

closed 2024-03-21 by xjh169