OpenIntelWireless/itlwm #972: Question regarding updating to new firmware

anonymous-writer, 2024-04-03:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: v2.3.0-alpha
- WiFi Card Model: AX-210
- PCI Product ID: -
- macOS Version: Sonoma

**Description**
I'm trying to install the latest firmware for my card with itlwm. The new kext load the firmware as it shoud be.
But I can't get to start scanning the network. The scan is simply canceled.

Do I have to additional things to get it work with new latest firmware ? 
I tested the firmware to see if it worked under Ubuntu and it works. Maybe someone knows what ist to do and want to help.

I did the update like it was done here.
https://github.com/OpenIntelWireless/itlwm/commit/54542db89d85890715418413a321c96a336c4eab

[Firmware_AX210.zip](https://github.com/OpenIntelWireless/itlwm/files/14851032/Firmware_AX210.zip)

Greetings Michael

zxystd, 2024-04-06:
Different with bluetooth firmwares, every wifi firmware has its features and APIs, we should do compat works on driver code, not simply loading the firmware into hardware. Why do you want to upgrade firmware? I had done some upgrade works for preparing support of wifi7 devices, but not really done and well testing so it is now only in my private repository.

anonymous-writer, 2024-04-06:
Thank you @zxystd  for your work.
I thought I could try this to see if anything improved in terms of compatibility. But with your answer I understood why it's not so easy.

I was playing around with Wireguard VPN yesterday and suddenly imassenger worked. Unfortunately I couldn't understand why. Activating VPN again didn't help.

Thanks again for your work.

closed 2024-04-06 by anonymous-writer

zxystd, 2024-04-07:
@anonymous-writer If you have another network interface to connect to the internet, then the imessage will work.

diepeterpan, 2024-04-07:
> @anonymous-writer If you have another network interface to connect to the internet, then the imessage will work.

Is there perhaps a loop-back / "ethernet proxy over wifi" that we can install to fake this?

anonymous-writer, 2024-04-08:
Yesterday I found a solution that makes imessage and facetime work again.
However, you need a second computer with an SSH server. A Rassbery, for example.

I'll still test and then write more.