OpenIntelWireless/itlwm #973: Sonoma 14.4.1 compatibility issue

Deftaudio, 2024-04-11:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.3.0-alpha
- WiFi Card Model: Intel AC-8265 integrated into NUC8i7HVK
- PCI Product ID: 0x1010
- macOS Version: 14.4.1

**Description**
I verified both AirportItlwm and itlwm and experience different problems but both of them are not working properly. I'm not sure if it's hardware related or Sonoma 14.4.1. is 14.4.1 different from 14.4 and requires another alpha build of kexts? 

1. AirportItlwm - looses the network, it shows that it's still connected though, however, the network traffic stops within several minutes after the start. Per log Log_2024-04-10_01-30-00.log it seems some watchdog timer kills all.
2. itlwm disconnects as well and sometime is causing system to crash. Log_2024-04-10_01-36-57.log 
[Log_2024-04-10_01-30-00.log](https://github.com/OpenIntelWireless/itlwm/files/14940678/Log_2024-04-10_01-30-00.log)
[Log_2024-04-10_01-36-57.log](https://github.com/OpenIntelWireless/itlwm/files/14940679/Log_2024-04-10_01-36-57.log)

**Bug Report Archive**
Log files attached. 

**Kext Download Source**
I only use precompiled version v2.3.0-alpha.

zxystd, 2024-04-11:
1. Are you using the latest alpha version?
2. Have you installed DebugEnhancer.kext, it seems that the first log file is incomplete.
3. If kernel panic happened, please attach the panic log.

Deftaudio, 2024-04-11:
1. yes, I'm using the latest 2.3.0-alpha posted. I haven't tried to compile from source. 
2. Sorry, it was disabled during that export that I realized later. I'll do that again and upload the data. 
3. I only saw kernel panic with itlwm. Would you prefer me focusing on this or continue investigating airportitlwm? 
Thank you.

zxystd, 2024-04-12:
panic log is already in your system. System info -> log, then find xxx.panic file, they were named with the panic date time.
[image]

zxystd, 2024-04-26:
@Deftaudio Hi?

Deftaudio, 2024-04-30:
Thank you for the follow up. 
I haven't checked itlwm more, but on AirportItlwm I think I found the problem. It seems related to going into sleep, when on a wake up I had to manually re-enable Wi-Fi.

zxystd, 2024-05-02:
> Thank you for the follow up. I haven't checked itlwm more, but on AirportItlwm I think I found the problem. It seems related to going into sleep, when on a wake up I had to manually re-enable Wi-Fi.

So it may be kind of sleep/wake issue, not related to this driver.

closed 2024-05-24 by zxystd