OpenIntelWireless/itlwm #975: Sonoma 14.5 panic when connected to certain wifi networks

johnpapad24, 2024-05-12:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES
**Environment**

- Kext Version: Airportitlwm v2.3.0-alpha (sonoma 14.4+) (4ac4c79)
- WiFi Card Model: Intel® Dual Band Wireless-AC 8260
- PCI Product ID: 
- macOS Version: sonoma 14.5 beta 1 to beta 4 (23F5074a) and possibly beyond

**Description**
panics experienced connected and downloading to certain wifi networks on airport itlwm and sonoma 14.5 beta using intel wireless ac 8260

The kernel panic is available below:

panic(cpu 3 caller 0xffffff8002d760a1): mbuf_watchdog: 1 waiters stuck for 25 secs
22867/22897 mbufs in use:
	22573 mbufs allocated to data
	215 mbufs allocated to packet headers
	79 mbufs allocated to socket names and addresses
	30 mbufs allocated to caches
659/682 mbuf 2KB clusters in use
13496/13503 mbuf 4KB clusters in use
0/0 mbuf 16KB clusters in use
61116 KB allocated to network (approx. 99% in use)
0 KB returned to the system
worker thread runs: 391, expansions: 209, cl 0/0, bigcl 390/69528, 16k 0/0
worker thread last run time: 298 (25 seconds ago)

top proc mbuf space 104192 bytes by wifip2pd:330

top ifcq_len 5 packets by en0

mbuf leak detection table:
	total captured: 1904 (one per 500)
	total allocs outstanding: 78
	new hash recorded: 1608 allocs, 585 traces
	hash collisions: 60 allocs, 286 traces
	overwrites: 1084 allocs, 0 traces
	lock conflicts: 0

top 5 outstanding traces:
[1] 25 outstanding alloc(s), 953 hit(s), 272 collision(s)
[2] 22 outstanding alloc(s), 22 hit(s), 0 collision(s)
[3] 9 outstanding alloc(s), 337 hit(s), 0 collision(s)
[4] 5 outstanding alloc(s), 21 hit(s), 0 collision(s)
[5] 4 outstanding alloc(s), 34 hit(s), 0 collision(s)

    trace [1]           trace [2]           trace [3]       	trace [4]           trace [5]      
    ------------------  ------------------  ------------------      ------------------  ------------------ 
 1: 0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  
 2: 0xffffff800094dc81  0xffffff800094dd28  0xffffff800094dc81  0xffffff800094dd28  0xffffff800094dd28  
 3: 0xffffff800097b490  0xffffff800097415e  0xffffff8000978eec  0xffffff8000978eec  0xffffff800097b490  
 4: 0xffffff8000946461  0xffffff800094dbf5  0xffffff800097bec7  0xffffff800097bec7  0xffffff8000946461  
 5: 0xffffff80025108cb  0xffffff800097b490  0xffffff8000945763  0xffffff8000945763  0xffffff80025108cb  
 6: 0x0  0xffffff8000946461  0x0  0x0  0x0  
 7: 0x0  0xffffff80025108cb  0x0  0x0  0x0  
 8: 0x0  0x0  0x0  0x0  0x0  
 9: 0x0  0x0  0xffffff8000b108aa  0xffffff8000b108aa  0x0  
10: 0xffffff8000b108aa  0x0  0xffffff8000b0f0ce  0xffffff8000b0f0ce  0xffffff8000b108aa  
11: 0xffffff8000b0f0ce  0x0  0xffffff8000b0e6f6  0xffffff8000b0e6f6  0xffffff8000b0f0ce  
12: 0xffffff8000b0e6f6  0xffffff8000b108aa                                          0xffffff8000b0e6f6  
13:                     0xffffff8000b0f0ce                                                              
14:                     0xffffff8000b0e6f6                                                              
15:                                                                                                     
16:                                                                                                     

Panicked task 0xffffff99af8f73d8: 4 threads: pid 1: launchd
[... 184 lines cut ...]
|IO80211!F	1200.13.0
>corecapture	1.0.4
|IOAHCIBlock!S	359
|IOAHCI!F	304
>!A!ILpssGspi	3.0.60
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
|IOSMBus!F	1.1
|IOTimeSync!F	1250.2
|IOSkywalk!F	1.0
>mDNSOffloadUserClient	1.0.1b8
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	9.0.0
|IOReport!F	47
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
>!ASSE

**Kext Download Source**
Github

johnpapad24, 2024-05-12:
panic(cpu 3 caller 0xffffff8002d760a1): mbuf_watchdog: 1 waiters stuck for 25 secs
22867/22897 mbufs in use:
	22573 mbufs allocated to data
	215 mbufs allocated to packet headers
	79 mbufs allocated to socket names and addresses
	30 mbufs allocated to caches
659/682 mbuf 2KB clusters in use
13496/13503 mbuf 4KB clusters in use
0/0 mbuf 16KB clusters in use
61116 KB allocated to network (approx. 99% in use)
0 KB returned to the system
worker thread runs: 391, expansions: 209, cl 0/0, bigcl 390/69528, 16k 0/0
worker thread last run time: 298 (25 seconds ago)

top proc mbuf space 104192 bytes by wifip2pd:330

top ifcq_len 5 packets by en0

mbuf leak detection table:
	total captured: 1904 (one per 500)
	total allocs outstanding: 78
	new hash recorded: 1608 allocs, 585 traces
	hash collisions: 60 allocs, 286 traces
	overwrites: 1084 allocs, 0 traces
	lock conflicts: 0

top 5 outstanding traces:
[1] 25 outstanding alloc(s), 953 hit(s), 272 collision(s)
[2] 22 outstanding alloc(s), 22 hit(s), 0 collision(s)
[3] 9 outstanding alloc(s), 337 hit(s), 0 collision(s)
[4] 5 outstanding alloc(s), 21 hit(s), 0 collision(s)
[5] 4 outstanding alloc(s), 34 hit(s), 0 collision(s)

    trace [1]           trace [2]           trace [3]       	trace [4]           trace [5]      
    ------------------  ------------------  ------------------      ------------------  ------------------ 
 1: 0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  0xffffff800097532a  
 2: 0xffffff800094dc81  0xffffff800094dd28  0xffffff800094dc81  0xffffff800094dd28  0xffffff800094dd28  
 3: 0xffffff800097b490  0xffffff800097415e  0xffffff8000978eec  0xffffff8000978eec  0xffffff800097b490  
 4: 0xffffff8000946461  0xffffff800094dbf5  0xffffff800097bec7  0xffffff800097bec7  0xffffff8000946461  
 5: 0xffffff80025108cb  0xffffff800097b490  0xffffff8000945763  0xffffff8000945763  0xffffff80025108cb  
 6: 0x0  0xffffff8000946461  0x0  0x0  0x0  
 7: 0x0  0xffffff80025108cb  0x0  0x0  0x0  
 8: 0x0  0x0  0x0  0x0  0x0  
 9: 0x0  0x0  0xffffff8000b108aa  0xffffff8000b108aa  0x0  
10: 0xffffff8000b108aa  0x0  0xffffff8000b0f0ce  0xffffff8000b0f0ce  0xffffff8000b108aa  
11: 0xffffff8000b0f0ce  0x0  0xffffff8000b0e6f6  0xffffff8000b0e6f6  0xffffff8000b0f0ce  
12: 0xffffff8000b0e6f6  0xffffff8000b108aa                                          0xffffff8000b0e6f6  
13:                     0xffffff8000b0f0ce                                                              
14:                     0xffffff8000b0e6f6                                                              
15:                                                                                                     
16:                                                                                                     

Panicked task 0xffffff99af8f73d8: 4 threads: pid 1: launchd
Backtrace (CPU 3), panicked thread: 0xffffff867e87eb30, Frame : Return Address
0xffffffc477e836d0 : 0xffffff8002826571 
0xffffffc477e83720 : 0xffffff800298bc00 
0xffffffc477e83760 : 0xffffff800297b1bc 
0xffffffc477e837e0 : 0xffffff80027bf971 
0xffffffc477e83800 : 0xffffff800282685d 
0xffffffc477e838f0 : 0xffffff8002825f03 
0xffffffc477e83950 : 0xffffff8002fdb38b 
0xffffffc477e83a40 : 0xffffff8002d760a1 
0xffffffc477e83a60 : 0xffffff8002d74818 
0xffffffc477e83ad0 : 0xffffff8002d4dbf5 
0xffffffc477e83b40 : 0xffffff8002d7415e 
0xffffffc477e83be0 : 0xffffff8002d4dbf5 
0xffffffc477e83c50 : 0xffffff8002d7a446 
0xffffffc477e83cc0 : 0xffffff8002d86fed 
0xffffffc477e83de0 : 0xffffff8002d983ee 
0xffffffc477e83ed0 : 0xffffff8002d980a1 
[... 164 lines cut ...]
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AXsanScheme	3
|IO80211!F	1200.13.0
>corecapture	1.0.4
|IOAHCIBlock!S	359
|IOAHCI!F	304
>!A!ILpssGspi	3.0.60
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
|IOSMBus!F	1.1
|IOTimeSync!F	1250.2
|IOSkywalk!F	1.0
>mDNSOffloadUserClient	1.0.1b8
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	9.0.0
|IOReport!F	47
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
>!ASSE

sameerasw, 2024-05-14:
Can confirm this behaviour for me but back in 14.2.1 .... Certain WiFi (Home WiFi) makes the entire OS instantly kernel panic and reboot.

Mostly I noticed when I turn off mobile hotspot so the WiFi networks automatically switches to home. It's just that one specific 2.4GHz WiFi so far.

zxystd, 2024-05-24:
Don't see anything relate to this kext, and no other people report the same issue.

zxystd, 2024-05-24:
@sameerasw If kernel panic happened, please get the panic log and report back to me, thanks.

closed 2024-05-24 by zxystd

johnpapad24, 2024-05-24:
the brand of the router with the problematic hotspot was huawei.