OpenIntelWireless/itlwm #976: [BUG] Itlwm.kext wont automatically connect to hidden wifi network

matte-oss, 2024-05-14:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 2.2.0
- WiFi Card Model: Intel dual band wireless-AC 7265
- PCI Product ID: 
- macOS Version: 12.7.4

**Description**

I add my hidden wifi network to HeliPort, everything works fine. I reboot my hackintosh and HeliPort has forgot my network, i need to re-add it
**Bug Report Archive**

I'm trying to install the debug kext, but It does not work
**Kext Download Source**

Official GitHub page

matte-oss, 2024-05-14:
Tried with my phone network (a wifi that isn't hidden) and It works normally

zxystd, 2024-05-24:
It is expected.

closed 2024-05-24 by zxystd

matte-oss, 2024-05-24:
Is there a way to make it connect automatically?

zxystd, 2024-05-26:
@matte-oss 
[image]
Hard code your ssid and password into the itlwm.kext/Contents/info.plist

matte-oss, 2024-05-26:
Thank you so much!

tiyo, 2026-01-05:
> [@matte-oss](https://github.com/matte-oss) [image] Hard code your ssid and password into the itlwm.kext/Contents/info.plist

Added, but it's still not working. Do I also need to set the password type?