OpenIntelWireless/itlwm #978: Ventura升级到13.6.7 AX201 突然断联重新联接后冻屏死机

quegfeng, 2024-05-25:
系统信息：

- Kext Version: 2.2.0
- WiFi Card Model: AX201
- PCI Product ID: 0X2725
- macOS Version: 13.6.7 amd RYzen平台

**Description**
用着好好的，突然没网，看wifi图标在联接，然后死机冻屏，只能强制断电重启。
kext load信息
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
  109    0 0xffffff8004639000 0xf51000   0xf51000   com.zxystd.AirportItlwm (2.2.0) 35A33FA2-8A65-3791-930C-96BB4734EAF6 <108 51 17 7 6 3 1>
第三方kext信息：
[
[image]
](url)

自己正常切换wifi没有故障
**Bug Report Archive**
**系统报告的问题：**
panic(cpu 2 caller 0xffffff80131e1475): zfree_drop: double free of 0xffffff9569957860 to zone kalloc.type.var3.32 @zalloc.c:815
Panicked task 0xffffff9f025a6e18: 220 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff95692c0b30, Frame : Return Address
0xffffffc494d5b6f0 : 0xffffff8012a70c7d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffc494d5b740 : 0xffffff8012bc5294 mach_kernel : _kdp_i386_trap + 0x114
0xffffffc494d5b780 : 0xffffff8012bb4da7 mach_kernel : _kernel_trap + 0x3b7
0xffffffc494d5b7d0 : 0xffffff8012a11971 mach_kernel : _return_from_trap + 0xc1
0xffffffc494d5b7f0 : 0xffffff8012a70f5d mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffc494d5b8e0 : 0xffffff8012a70607 mach_kernel : _panic_trap_to_debugger + 0x1a7
0xffffffc494d5b940 : 0xffffff80131db8eb mach_kernel : _panic + 0x84
0xffffffc494d5ba30 : 0xffffff80131e1475 mach_kernel : ___smr_tail_invalid + 0x294f
0xffffffc494d5ba60 : 0xffffff8012ae1433 mach_kernel : _workload_config_available + 0x8063
0xffffffc494d5ba90 : 0xffffff80130d0e40 mach_kernel : _IOFree + 0x20
0xffffffc494d5bab0 : 0xffffff8016c8e86b com.zxystd.AirportItlwm : __ZL4freePv + 0x4b
0xffffffc494d5bae0 : 0xffffff8016c97054 com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_send_cmdEP9iwx_softcP12iwx_host_cmd + 0x814
0xffffffc494d5bc00 : 0xffffff8016c9f49c com.zxystd.AirportItlwm : __ZN6ItlIwx19iwx_send_cmd_statusEP9iwx_softcP12iwx_host_cmdPj + 0x4c
0xffffffc494d5bc60 : 0xffffff8016c995c1 com.zxystd.AirportItlwm : __ZN6ItlIwx23iwx_send_cmd_pdu_statusEP9iwx_softcjtPKvPj + 0x71
0xffffffc494d5bcf0 : 0xffffff8016c99282 com.zxystd.AirportItlwm : __ZN6ItlIwx14iwx_sta_rx_aggEP9iwx_softcP14ieee80211_nodehttii + 0x182
0xffffffc494d5bdb0 : 0xffffff8016c9a70c com.zxystd.AirportItlwm : __ZN6ItlIwx11iwx_ba_taskEPv + 0x16c
0xffffffc494d5be40 : 0xffffff8016c3ccef com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x6f
0xffffffc494d5bfa0 : 0xffffff8012a1119e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.2)[35A33FA2-8A65-3791-930C-96BB4734EAF6]@0xffffff8016c39000->0xffffff8017b7ffff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[DA9C6809-0196-35CE-AA17-063A51D5A68B]@0xffffff80146f4000->0xffffff801483afff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[9464DB11-7BAB-372E-BE42-312E2C708040]@0xffffff8015254000->0xffffff801526afff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[05751018-D5D6-39CE-9F45-B2652D5B9E94]@0xffffff80154b5000->0xffffff80154e6fff

Process name corresponding to current thread (0xffffff95692c0b30): kernel_task
Boot args: -v keepsyms=1 debug=0x144 alcid=28 alctsel=1 watchdog=0 -NRedDPDelay -wegnoegpu revblock=media -vi2c-force-polling -lilubetaall 

Mac OS version:
22G720

Kernel version:
Darwin Kernel Version 22.6.0: Mon Apr 22 20:54:28 PDT 2024; root:xnu-8796.141.3.705.2~1/RELEASE_X86_64
Kernel UUID: FC979E73-CAD7-35BF-812E-E4B3C923BA96
roots installed: 0
KernelCache slide: 0x0000000012600000
KernelCache base:  0xffffff8012800000
Kernel slide:      0x00000000126dc000
Kernel text base:  0xffffff80128dc000
__HIB  text base: 0xffffff8012700000
System model name: MacBookPro16,3 (Mac-E7203C0F68AA0004)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 1740625031122
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000019545563420
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000f1c351220 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8a34ea3000 - 0xffffffaa34ea3000
  . PGZ   : 0xffffff8a34ea3000 - 0xffffff8a36ea4000
  . VM    : 0xffffff8a36ea4000 - 0xffffff8f036a3000
  . RO    : 0xffffff8f036a3000 - 0xffffff909cea3000
  . GEN0  : 0xffffff909cea3000 - 0xffffff95696a3000
  . GEN1  : 0xffffff95696a3000 - 0xffffff9a35ea3000
  . GEN2  : 0xffffff9a35ea3000 - 0xffffff9f026a3000
  . GEN3  : 0xffffff9f026a3000 - 0xffffffa3ceea3000
  . DATA  : 0xffffffa3ceea3000 - 0xffffffaa34ea3000
  Metadata: 0xffffff8031413000 - 0xffffff8051413000
  Bitmaps : 0xffffff8051413000 - 0xffffff8054413000
  Extra   : 0 - 0

已复现三次，如重现还要其他信息，下次加上。

zxystd, 2024-05-26:
试试2.3.0-alpha

quegfeng, 2024-05-31:
> 试试2.3.0-alpha

ok,暂时没出现这种情况了。

closed 2024-06-09 by zxystd