OpenIntelWireless/itlwm #979: Randomly USB(XHCI) disabled on Sonoma 14.5

huang207, 2024-05-27:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.3.0-alpha-4ac4c79
- WiFi Card Model: Intel AC9560
- PCI Product ID: 808634F0
- macOS Version: Sonoma 14.5

**Description**
Randomly USB disabled when loaded AirPortitlwm.

**Bug Report Archive**

normal means USB work well on this boot
[bugreport_4800_disabled.zip](https://github.com/OpenIntelWireless/itlwm/files/15455513/bugreport_4800_disabled.zip)
[bugreport_64496_normal.zip](https://github.com/OpenIntelWireless/itlwm/files/15455515/bugreport_64496_normal.zip)

**Kext Download Source**

Github 2.3.0-alpha-4ac4c79

huang207, 2024-05-27:
[image]
Only the 3rd boot work well.
An IOUSBHostDevice can be found on the 2nd, and no device found on the 1st

zxystd, 2024-05-27:
Not relate.

closed 2024-05-27 by zxystd