OpenIntelWireless/itlwm #981: Airportitlwm and SecureBootModel

bhuether, 2024-06-08:
Hi, I have seen in issues it discussed, that airportitlwm works best with SecureBootModel disabled. But for Mac updates to work, one needs this as Default. Trying to understand how to reconcile these requirements. On Sonoma 14.5, so just trying to figure out what is currently suggested for stability and compatibility. Thanks.

Sorry if this isn't the right forum for this sort of question.

zxystd, 2024-06-08:
> that airportitlwm works best with SecureBootModel disabled

How do you make this conclusion?

closed 2024-06-08 by zxystd

bhuether, 2024-06-08:
Hi, actually, I have it the other way around, that best as Default, but
that Disabled is needed in other cases for certain aspects of OpenCore.  So
I was wondering if there ever is a time I set SecureBootMode to Disabled,
will that have impact on Airportitwlm?

thanks!