OpenIntelWireless/itlwm #983: Add support for macos 15 Sequoia Beta

Lorys89, 2024-06-11:
In the first tests, itlwm and heliport work

perez987, 2024-06-11:
Yes, itlwm and Heliport work for me too, AirportItlwm can't enable wifi.

ankanpratik, 2024-06-11:
> So, can you guys use wifi? I can't turn on wifi on neither airportitlwm and itlwm

If you're using itwlm, make sure to install Heliport app from OpenIntelWireless repo. That way you can find and connect through wifi and it works perfectly. Won't work natively through the MacOS wifi setting with itlwm

perez987, 2024-06-11:
@ruanyouxing 
Yes, I have wifi with itlwm.kext + Heliport.
@ankanpratik 
I use a [Heliport fork by diepeterpan](https://github.com/diepeterpan/HeliPort), it has slight enhancements for Sonoma.

perez987, 2024-06-12:
@ruanyouxing 
Remember differences between the 2 methods:

* `itlwm.kext`: uses `IOEthernetController` instead of `IO80211Family` so the connection spoofs as Ethernet even though it works as wifi. It does not use the macOS Wi-Fi menu, instead you have to use the HeliPort application. On Ventura you can have itlwm_v2.2.0_stable.kext or itlwm_v2.3.0_stable.kext. On Sonoma you need itlwm_v2.3.0_stable.kext.
* `AirportItlwm.kext`: uses `IO80211Family` so it works like the rest of the system's Wi-Fi connections. It provides minimal Continuity features (Handoff and Universal Clipboard, not always available) but cannot connect to hidden networks. No HeliPort needed. On Ventura you need AirportItlwm_v2.3.0_stable_Ventura.kext.zip. On Sonoma prior to 14.4 you need AirportItlwm_v2.3.0_stable_Sonoma14.0. On Sonoma 14.4+ you need AirportItlwm_v2.3.0_stable_Sonoma14.4.

johnpapad24, 2024-06-13:
Itlwm (2.3.0) with heliport seems to be working well in my t460 (intel 8260) using sequoia beta1.

moreover the bluetooth also seems to be working (without intelbtpatcher kext)

shoxrux2006, 2024-06-13:
AirportItlwm Not working on sequia

perez987, 2024-06-14:
> AirportItlwm Not working on sequia

Not on my system either.

JoaoVitor42, 2024-06-17:
@Lorys89 can you share de .kext for local test?

fangshaok, 2024-06-21:
我使用 itlwm+heliport 无法工作，我创建了诊断报告如下：
`[    1.291182]: itlwm: iwm_match devId=0x5910
[    1.291195]: itlwm: free
[    1.291314]: itlwm: iwm_match devId=0x1901
[    1.291320]: itlwm: iwm_match devId=0xA110
[    1.291320]: itlwm: free
[    1.291328]: itlwm: free
[    1.291574]: itlwm: iwm_match devId=0x1903
[    1.291579]: itlwm: free
[    1.291631]: itlwm: iwm_match devId=0xA116
[    1.291638]: itlwm: free
[    1.294572]: itlwm: iwm_match devId=0xA131
[    1.294583]: itlwm: free
[    1.295487]: itlwm: iwm_match devId=0xA12F
[    1.295495]: itlwm: free
[    1.295699]: itlwm: iwm_match devId=0xA13A
[    1.295714]: itlwm: free
[    1.295872]: itlwm: iwm_match devId=0xA118
[    1.295878]: itlwm: free
[    1.296767]: itlwm: iwm_match devId=0xA152
[    1.296779]: itlwm: free
[    1.296969]: itlwm: iwm_match devId=0xA115
[    1.296983]: itlwm: free
[    1.303861]: itlwm: iwm_match devId=0xA103
[    1.303867]: itlwm: free
[    1.304120]: itlwm: iwm_match devId=0x095A
[    1.306078]: itlwm: iwm_match devId=0x591B
[    1.306089]: itlwm: free
[    1.306385]: itlwm: iwm_match devId=0xA171
[    1.306402]: itlwm: free
[    1.306449]: itlwm: iwm_match devId=0xA114
[    1.306460]: itlwm: free
[    1.306751]: itlwm: iwm_match devId=0xA123
[    1.306765]: itlwm: free
[    1.309784]: itlwm: iwm_match devId=0xA121
[    1.309794]: itlwm: free
[    1.507272]: itlwm: iwm_attach
[    1.509290]: itlwm: alloc contig
[    1.625653]: itlwm: config ieee80211
[    1.625661]: itlwm: ieee80211_channel_init
[    1.625666]: itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
[    1.625672]: itlwm: ieee80211_media_init
[    1.625705]: itlwm: iwm_preinit
[    1.625707]: itlwm: iwm_prepare_card_hw
[    1.625730]: itlwm: taskq taskq_create_thread lock
[    1.625734]: itlwm: taskq taskq_create_thread unlock
[    1.625740]: itlwm: taskq taskq_create_thread lock
[    1.625741]: itlwm: taskq taskq_create_thread unlock
[    1.625746]: itlwm: taskq taskq_create_thread lock
[    1.625747]: itlwm: taskq taskq_create_thread unlock
[    1.625750]: itlwm: taskq taskq_create_thread lock
[    1.625752]: itlwm: taskq taskq_create_thread unlock
[    1.625754]: itlwm: taskq taskq_thread schedule task
[    1.625757]: itlwm: taskq taskq_thread schedule task
[    1.625918]: itlwm: iwm_start_hw
[    1.625919]: itlwm: iwm_prepare_card_hw
[    1.631402]: itlwm: iwm_apm_init
[    1.631892]: itlwm: itlwm: L1 Disabled - LTR Enabled
[    1.632029]: itlwm: iwm_init_msix_hw
[    1.632030]: itlwm: iwm_conf_msix_hw
[    1.632038]: itlwm: iwm_check_rfkill RF_KILL hw: 1
[    1.632040]: itlwm: iwm_run_init_mvm_ucode justnvm=1
[    1.632042]: itlwm: iwm_load_ucode_wait_alive ucode_type=1
[    1.637903]: itlwm: load firmware iwm-7265-17 done
[    1.638025]: itlwm: iwm_start_fw ucode_type=1
[    1.638027]: itlwm: iwm_apm_init
[    1.638062]: itlwm: itlwm: L1 Disabled - LTR Enabled
[    1.638300]: itlwm: iwm_load_firmware
[    1.660053]: itlwm: iwm_post_alive
[    1.660395]: itlwm: iwm_enable_ac_txq
[    1.660458]: itlwm: iwm_send_bt_init_conf
[    1.662861]: itlwm: iwm_nvm_init
[    1.662879]: itlwm: iwm_nvm_read_section
[    1.663008]: itlwm: iwm_nvm_read_section
[    1.663084]: itlwm: iwm_nvm_read_section
[    1.663133]: itlwm: iwm_nvm_read_section
[    1.663244]: itlwm: iwm_nvm_read_section
[    1.663371]: itlwm: iwm_nvm_read_section
[    1.663504]: itlwm: iwm_nvm_read_section
[    1.663549]: itlwm: iwm_nvm_read_section
[    1.663588]: itlwm: iwm_nvm_read_section
[    1.663654]: itlwm: iwm_parse_nvm_sections
[    1.663659]: itlwm: iwm_stop_device
[    1.700600]: itlwm: iwm_conf_msix_hw
[    1.700610]: itlwm: iwm_check_rfkill RF_KILL hw: 1
[    1.700612]: itlwm: iwm_prepare_card_hw
[    1.700627]: itlwm: itlwm: hw rev 0x210, fw ver 17.3216344376.0, address d0:57:7b:ed:c9:cf
[    1.700630]: itlwm: ieee80211_channel_init
[    1.700634]: itlwm: ieee80211_media_init
[    1.700654]: itlwm: attach succeed.
[   16.616757]: process_is_plugin_host: running binary "bash" in keys-off mode due to identity: com.apple.bashprocess_is_plugin_host: running binary "bash" in keys-off mode due to identity: com.apple.bashitlwm: enable
[   16.725340]: itlwm: virtual IOReturn ItlIwm::enable(IONetworkInterface *)
[   16.725342]: itlwm: iwm_resume
[   16.725360]: itlwm: iwm_check_rfkill RF_KILL hw: 1
[   16.725362]: itlwm: iwm_prepare_card_hw
[   16.725384]: itlwm: iwm_init_task
[   19.031726]: itlwm: iwm_match devId=0xA171
[   19.031736]: itlwm: free
[   19.037733]: itlwm: iwm_match devId=0xA121
[   19.037750]: itlwm: free
[   19.049632]: itlwm: iwm_match devId=0xA123
[   19.049643]: itlwm: free
[   19.233213]: itlwm: iwm_match devId=0x591B
[   19.233218]: itlwm: free

Time ran: 2024-06-21 08:52:30.6850
itlwm loaded version: 2.3.0 (Firmware: 17.3216344376.0)

macOS 版本15.0（版号24A5264n）`

Lorys89, 2024-06-22:
@zxystd  do you have news for us? Is it possible to run airportitlwm on sequoia?

rudo56, 2024-06-22:
No, AirportItlwm_v2.3.0_stable_Sonoma14.4 not working on latest sonoma 14.5 and Sequoia Beta

sweetpotato133, 2024-06-22:
> No, AirportItlwm_v2.3.0_stable_Sonoma14.4 not working on latest sonoma 14.5 and Sequoia Beta

It's working on Sonoma 14.5 but not on Sequoia beta...

Lorys89, 2024-06-23:
I'm doing some tests, It still doesn't work 🤯🙈
[image]
[image]

rudo56, 2024-06-23:
> > No, AirportItlwm_v2.3.0_stable_Sonoma14.4 not working on latest sonoma 14.5 and Sequoia Beta
> 
> It's working on Sonoma 14.5 but not on Sequoia beta...

No iservices

zxystd, 2024-06-24:
Hi, I am away from Hackintosh machine for a few months, and don't have such spare time to look into this update. I had done  IO80211Family headers update, but the implement is changed, Apple put their wifi driver to user space now, and route IOCTL commands through UserClient to kernel directly, I am finding ways to bypass this, since AirportItlwm is running in the kernel side. But the most important thing is that I don't have machine at hand :(

maxpicelli, 2024-06-24:
Hi, Its possible we can help you to get an machine to work on that ?

zxystd, 2024-06-24:
> I'm doing some tests, It still doesn't work 🤯🙈 ![Screenshot_2024-06-23_alle_07 24 46](https://private-user-images.githubusercontent.com/64613690/342029719-287e60c5-f64f-42fa-9e8e-a979d0c7f5e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyMDQ0NjIsIm5iZiI6MTcxOTIwNDE2MiwicGF0aCI6Ii82NDYxMzY5MC8zNDIwMjk3MTktMjg3ZTYwYzUtZjY0Zi00MmZhLTllOGUtYTk3OWQwYzdmNWU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDA0NDI0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxZTc0YjQ0YzA0NmI4ZWJkYmVhZjA3NzRhZTViYmYzZjE4ZjkwMmM2MDY2NWUwYWY2MjM2NzQyOWI0OTU5MDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0Ad0YOIKVCGB7RF_Ujj2ICzsLgFssZPMar_VDk0MFUs) ![Screenshot_2024-06-23_alle_07 15 07](https://private-user-images.githubusercontent.com/64613690/342029725-eb99a331-af51-4020-9450-01336356aabf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyMDQ0NjIsIm5iZiI6MTcxOTIwNDE2MiwicGF0aCI6Ii82NDYxMzY5MC8zNDIwMjk3MjUtZWI5OWEzMzEtYWY1MS00MDIwLTk0NTAtMDEzMzYzNTZhYWJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDA0NDI0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ0YTE2NzAzZmVlMTFlYTQxNmRiNzk4ZDAyZGRmMTkyMDNiMzY0YjA1MjhkZTM5ZmE5ZWM5ZDViY2FhMjE1YWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7HUZ982-OvmSpR-gkflJzDkuYZjYRKIiLkfJG7SrFdo)
> 
> [Log_2024-06-23_08-52-57.log](https://github.com/user-attachments/files/15942866/Log_2024-06-23_08-52-57.log)

I think you get further progress than me at this moment, have you already update the headers?

zxystd, 2024-06-24:
> Hi, Its possible we can help you to get an machine to work on that ?
> 
> You brilliant work is very important to Hackintosh World. Please let is help you from some way.

Thank you, I have machines, but I don't have much spare time in short to touch them and upgrade to macOS 15.

Lorys89, 2024-06-24:
> I think you get further progress than me at this moment, have you already update the headers?
No, and I'm at a standstill. I'm waiting for you.. It's too big a project for me. 🙈
When you're freer, you'll fix it.

Shaneee, 2024-06-24:
> I think you get further progress than me at this moment, have you already update the headers?

Is it possible you could create a branch with these changes? You've made more progress than I've made. I'd like to contribute. 

This is as far as I got lol https://cdn.discordapp.com/attachments/1041470332541878282/1254374160696610836/Screenshot_2024-06-23_at_10.54.13.png?ex=667a93f9&is=66794279&hm=38db8f73644d0f3a7cbf98791d9aee8c31dde44027e56fdf3bc5b1754ffeaa6c&

anonymous-writer, 2024-06-25:
Beta 2 "iPhone Mirroring" maybe only work with  AirportItlwm.

[image]

A pity.

perez987, 2024-06-25:
@anonymous-writer 
And iOS 18.

lwx1901, 2024-06-26:
期待Intel Wi-Fi早日实现：

[image]

davidereitano, 2024-07-02:
same issue here.
itlwm + heliport works, but since AirportItlwm can't enable wifi, for the system there's no wifi connection.
iMessage, continuity and other internet related things works but (for example) sidecar or universal control don't work anymore

slose1, 2024-07-10:
[image]
i patched like bcm94360
kext IOSkywalkFamily 1.1.0, IO80211FamilyLegacy and block com.apple.iokit.IOSkywalkFamily
and add Ventura Airportitlwm.kext
It doesn't seem to recognize wifi

maxpicelli, 2024-07-10:
> ![image](https://private-user-images.githubusercontent.com/169359119/347346997-f837f7d8-3a52-4200-aae8-8f0af96bbf4d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

I guess the OCLP configuration not fix recognize intel wireless. It's only for Broadcom series.

Shaneee, 2024-07-17:
Think I'll wait for [zxystd](https://github.com/zxystd) to update this. I'm pulling my hair out lol. I'm getting panics relating to AirportItlwmEthernetInterface and can't for the life of me get it sorted...

Daasin, 2024-07-25:
That's a shame, if there's any inprocess code people need me to try. I'd be more than happy to test on sequoia across multiple machines.

Main one that i'd spend most time on has BE200 though

Serverbd-Technology, 2024-07-31:
@zxystd Please bro update this ASAP We are waiting for your great work..

macsupermac, 2024-08-11:
So,is there anything progress?Thanks!

Daasin, 2024-08-11:
> So,is there anything progress?Thanks!

_"Hi, I am away from Hackintosh machine for a few months, and don't have such spare time to look into this update."_ - Was the last update, so it looks like it'll atleast be a few months.

davidereitano, 2024-08-12:
anyone knows if there's any wifi/bt card that works OOB with sequoia?

maxpicelli, 2024-08-12:
> anyone knows if there's any wifi/bt card that works OOB with sequoia?

Its better you try OCLP patcher because at moment not exist any comercial wireless available to work OOB on Sequoia.

Zormeister, 2024-08-24:
> Hi, I am away from Hackintosh machine for a few months, and don't have such spare time to look into this update. I had done IO80211Family headers update, but the implement is changed, Apple put [...]

let me know if you need a helping hand at all on the Sequoia API.

Daasin, 2024-08-24:
> > Hi, I am away from Hackintosh machine for a few months, and don't have such spare time to look into this update. I had done IO80211Family headers update, but the implement is changed, Apple put their wifi driver to user space now, and route IOCTL commands through UserClient to kernel directly, I am finding ways to bypass this, since AirportItlwm is running in the kernel side. But the most important thing is that I don't have machine at hand :(
> 
> let me know if you need a helping hand at all on the Sequoia API, I’m willing to help with what I know about IOSkywalk and IO80211

Maybe @Shaneee could use the help on figuring out the crash? Since zxystd is currently away atm and again, I'm happy to test on multiple builds if needed.

Serverbd-Technology, 2024-08-25:
> Beta 2 "iPhone Mirroring" maybe only work with AirportItlwm.
> 
> ![3b9bebd8-8d84-4dcb-83da-7102e5153aea](https://private-user-images.githubusercontent.com/35840715/342603283-2494bb00-f2d7-4e1a-ae24-b638387c37fe.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1OTAzMTAsIm5iZiI6MTcyNDU5MDAxMCwicGF0aCI6Ii8zNTg0MDcxNS8zNDI2MDMyODMtMjQ5NGJiMDAtZjJkNy00ZTFhLWFlMjQtYjYzODM4N2MzN2ZlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI1VDEyNDY1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2MDU3MjQzNzE4ZjQzNzBkNWU5NmJiZTRjYmYzZGUyODBjYjdjZjFjMjQ5ZGQ1YzkyYTVlMjgzNTM1NjdlZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Sf8m7Gly7IzZKZCEGS8mCceGbpQ9-NTKtSU7pokSk4s)
> 
> Too bad.
Need  A Mac with an Apple Silicon chip or a T2 Security Chip for Mirroring iPhone.

chteow, 2024-08-26:
> let me know if you need a helping hand at all on the Sequoia API, I’m willing to help with what I know about IOSkywalk and IO80211

saw @Lorys89 working on https://github.com/Lorys89/itlwm test2 but still fail to bring up the wifi. 
Maybe you can take a look from there. I'm not in 'C' so not able to help.

perez987, 2024-08-26:
Yes, @Lorys89 has worked on this but it's not valid yet. In fact, he is the creator of this issue.

chteow, 2024-08-27:
> If it doesn't crash the machine on boot up I'll try that

it won't crash, Mac bootup, just airportitlwm not loaded but on my machine.

Zormeister, 2024-08-30:
> Think I'll wait for [zxystd](https://github.com/zxystd) to update this. I'm pulling my hair out lol. I'm getting panics relating to AirportItlwmEthernetInterface and can't for the life of me get it sorted...

i suggest that AirportItlwmEthernetInterface get scrapped entirely and replaced with a non-stub implementation of IOSkywalkEthernetInterface subclass (aka: proper implementation of the Skywalk API)

of course, that would need a rewrite of AirportItlwm, though.

TommyLuco, 2024-09-13:
> > > Think I'll wait for [zxystd](https://github.com/zxystd) to update this. I'm pulling my hair out lol. I'm getting panics relating to AirportItlwmEthernetInterface and can't for the life of me get it sorted...
> > 
> > 
> > i suggest that AirportItlwmEthernetInterface get scrapped entirely and replaced with a non-stub implementation of an IOSkywalkEthernetInterface subclass (aka: proper implementation of the Skywalk API/IO80211SkywalkInterface)
> > of course, that would need a rewrite of AirportItlwm, though.
> 
> why not you help to make something out here, seem like you are expertise as well. Thank in advances.

A complete rewrite isn’t just done easy peasy. Giving technical pointers is a completely different thing so just requesting from some random person on the internet to just do it themselves just because they understand at least to a degree some technical limitations of the current code is honestly insane…

Zormeister, 2024-09-13:
the way itlwm is currently written, from my own observation, is that AirportItlwm feeds IO80211Family just enough info to work, relying on mac80211 from BSD to work internally, what I suggest however is the complete integration of AirportItlwm with IO80211Family, but that isn’t an easy task.

bulyaki, 2024-09-18:
Hi, I'm using Itlwm with Heliport on Sequoia 15.0 final, and I am also getting those random resets, kernel panic caused by itlwm. Here is my error report:
--------------------------------------------
```
panic(cpu 0 caller 0xffffff8003fda557): zfree_drop: double free of 0xffffff8cee38fac0 to zone kalloc.type.var5.32 @zalloc.c:830
Panicked task 0xffffff9680a5e870: 4601 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff91b6850b30, Frame : Return Address
0xffffffd75235f680 : 0xffffff80037ecc41 mach_kernel : _handle_debugger_trap + 0x4c1
0xffffffd75235f6d0 : 0xffffff80039598ec mach_kernel : _kdp_i386_trap + 0x11c
0xffffffd75235f710 : 0xffffff8003948f6b mach_kernel : _kernel_trap + 0x48b
0xffffffd75235f7d0 : 0xffffff8003782971 mach_kernel : _return_from_trap + 0xc1
0xffffffd75235f7f0 : 0xffffff80037ecf37 mach_kernel : _DebuggerTrapWithState + 0x67
0xffffffd75235f8f0 : 0xffffff80037ec5d2 mach_kernel : _panic_trap_to_debugger + 0x1e2
0xffffffd75235f960 : 0xffffff8003fd4938 mach_kernel : _panic + 0x81
0xffffffd75235fa50 : 0xffffff8003fda557 mach_kernel : ___smr_stail_invalid + 0x26a1
0xffffffd75235fa80 : 0xffffff8003863575 mach_kernel : _workload_config_available + 0x4465
0xffffffd75235fab0 : 0xffffff8003eb2520 mach_kernel : _IOFree + 0x20
0xffffffd75235fad0 : 0xffffff80081c8274 com.zxystd.itlwm : __ZL4freePv + 0x54
0xffffffd75235fb00 : 0xffffff80081d0cd8 com.zxystd.itlwm : __ZN6ItlIwx12iwx_send_cmdEP9iwx_softcP12iwx_host_cmd + 0x868
0xffffffd75235fc10 : 0xffffff80081d94d4 com.zxystd.itlwm : __ZN6ItlIwx19iwx_send_cmd_statusEP9iwx_softcP12iwx_host_cmdPj + 0x64
0xffffffd75235fc70 : 0xffffff80081d33d5 com.zxystd.itlwm : __ZN6ItlIwx23iwx_send_cmd_pdu_statusEP9iwx_softcjtPKvPj + 0x75
0xffffffd75235fcf0 : 0xffffff80081d308e com.zxystd.itlwm : __ZN6ItlIwx14iwx_sta_rx_aggEP9iwx_softcP14ieee80211_nodehttii + 0x1ee
0xffffffd75235fdc0 : 0xffffff80081d4649 com.zxystd.itlwm : __ZN6ItlIwx11iwx_ba_taskEPv + 0x1b9
0xffffffd75235fe40 : 0xffffff8008257407 com.zxystd.itlwm : __Z12taskq_threadPv + 0x97
0xffffffd75235ffa0 : 0xffffff800378219e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(2.3)[3BDD2D98-5A67-3154-95DE-1F394C8365C3]@0xffffff80081c5000->0xffffff8009142fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[FCB9BFDD-93F0-3784-8274-3D4DB73BF90E]@0xffffff8005998000->0xffffff80059adfff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[7EAD85CE-638E-3217-A14D-09DF66BA8265]@0xffffff8005bfa000->0xffffff8005c2cfff

Process name corresponding to current thread (0xffffff91b6850b30): kernel_task
Boot args: -v keepsyms=1 debug=0x100 alcid=1 amfi=0x80 vsmcgen=1 

Mac OS version:
24A335

Kernel version:
Darwin Kernel Version 24.0.0: Mon Aug 12 20:54:30 PDT 2024; root:xnu-11215.1.10~2/RELEASE_X86_64
Kernel UUID: 5DD51D41-0315-3DDD-BD5D-50E782643BDB
roots installed: 0
KernelCache slide: 0x0000000003400000
KernelCache base:  0xffffff8003600000
Kernel slide:      0x00000000034e4000
Kernel text base:  0xffffff80036e4000
__HIB  text base: 0xffffff8003500000
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 72704506395458
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000421fd646fece
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000002ecf7cf4f1 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff86789a6000 - 0xffffffa6789a6000
  . PGZ   : 0xffffff86789a6000 - 0xffffff86889a9000
  . VM    : 0xffffff86889a9000 - 0xffffff8b5300e000
  . RO    : 0xffffff8b5300e000 - 0xffffff8cebcda000
  . GEN0  : 0xffffff8cebcda000 - 0xffffff91b6340000
  . GEN1  : 0xffffff91b6340000 - 0xffffff96809a6000
  . GEN2  : 0xffffff96809a6000 - 0xffffff9b4b00c000
  . GEN3  : 0xffffff9b4b00c000 - 0xffffffa015672000
  . DATA  : 0xffffffa015672000 - 0xffffffa6789a6000
  Metadata: 0xffffffa69c9b6000 - 0xffffffa6bc9b6000
  Bitmaps : 0xffffffa6bc9b6000 - 0xffffffa6cc9b6000
  Extra   : 0 - 0
```
--------------------------------------------

zxystd, 2024-09-22:
@bulyaki 
[itlwm.kext.zip](https://github.com/user-attachments/files/17086247/itlwm.kext.zip)
please try this one to see if it can help?

bulyaki, 2024-09-22:
Hi @zxystd, thanks for the help. I updated my itlwm.kext, and the system failed to boot with the new one, it just kept rebooting even before it loaded the window manager. I faintly remembered a similar issue so I disabled SIP to see if it boots, and it did. Unfortunately this is not a desirable configuration for me, I need the SIP enabled because some of my apps are not working otherwise. E.g. I can't play any video with Stremio and other players, they just give me a black screen and the logs keep reporting some sort of weird Qt Webkit error, but everything works just fine if I enable SIP. 
So in the end I restored my old kext and rebooted with SIP.

Below is the kernel panic from the boot-up, it seems it has some sort of issue with the signature of the kext maybe? My csr-active-config is set to "00080000", I am not sure what these flags actually mean, but is there a chance that I could still load your kext and boot with SIP enabled if I modified some of these flags?

```
----------------------------------------------
panic(cpu 0 caller 0xffffff80050f2c29):  initproc failed to start -- exit reason namespace 6 subcode 0x1 description: Library not loaded: /System/Library/PrivateFrameworks/IO80211.framework/Versions/A/IO80211
  Referenced from: <2E9E97B3-4E8C-32B0-986B-EFEFAD003B12> /System/Library/PrivateFrameworks/CoreUtils.framework/Versions/A/CoreUtils
  Reason: tried: '/System/Library/PrivateFrameworks/IO80211.framework/Versions/A/IO80211' (code signature invalid in <C0791260-3DB4-30E8-962C-1D7A90BBE2D5> '/System/Library/PrivateFrameworks/IO80211.framework/Versions/A/IO80211' (errno=1) sliceOffset=0x00000000, codeBlobOffset=0x00003DF0, codeBlobSize=0x000047F0), '/System/Volumes/Preboot/Cryptexes/OS/System/Library/PrivateFrameworks/IO80211.framework/Versions/A/IO80211' (no such file), '/System/Library/PrivateFrameworks/IO80211.framework/Versions/A/IO80211' (code signature invalid in <C0791260-3DB4-30E8-962C-1D7A90BBE2

uuid info:
 0x10b333000	uuid = <e6056c94-fc2d-3517-b1e1-46d8eb58a10e>
 0x104a6a000	uuid = <163ac458-14da-3067-87a1-5c02b7612a75>

Thread 0 crashed

RAX: 0x0000000002000209, RBX: 0x0000000000000001, RCX: 0x00007ff7bb483d98, RDX: 0x00007ff7bb484200
RSP: 0x00007ff7bb483d98, RBP: 0x00007ff7bb483de0, RSI: 0x0000000000000001, RDI: 0x0000000000000006
R8:  0x00007ff7bb483e00, R9:  0x0000000000000000, R10: 0x00000000000000a6, R11: 0x0000000000000246
R12: 0x0000000000000000, R13: 0x00007ff7bb484200, R14: 0x0000000000000006, R15: 0x00000000000000a6
RFL: 0x0000000000000246, RIP: 0x000000010b393b16, CS:  0x0000000000000007, SS:  0x0000000000000023

Thread 0: 0xffffff9a326320c8
	0x000000010b393b16
	0x000000010b3ab929
	0x000000010b33d7a1
	0x000000010b33a822
	0x000000010b339792
	0x000000010b33927f

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 24.0.0: Mon Aug 12 20:54:30 PDT 2024; root:xnu-11215.1.10~2/RELEASE_X86_64
Kernel UUID: 5DD51D41-0315-3DDD-BD5D-50E782643BDB
roots installed: 0
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization
----------------------------------------------
```

zxystd, 2024-09-22:
itlwm.kext doesnt need to request IO80211Framework, it should not caused by itlwm. please check your setting?

bulyaki, 2024-09-22:
The system's IO80211Family framework is needed by the AppleEthernetMLX5 driver, because I have a Mellanox card installed. But somehow it must conflict with the itlwm driver, because as soon as I switched back to AirportItlwm (which does not work but at least does not crash either) suddenly I was able to finish the boot process. When I switched to the new itlwm the only change I made was that I disabled AirportItlwm and enabled itlwm, I didn't touch anything else.

Zormeister, 2024-09-22:
did you OCLP your system

bulyaki, 2024-09-22:
> yes, you did. the reason for the crash is because the code signature is invalid, meaning you need to neutralise AMFI (eg, through AMFIPass) and obliterate SIP more (car-active-config `0A08000000`)

Yes I did, should I have not done that? Thanks, I try this now.

bulyaki, 2024-09-22:
Great, it boots now, window server starts, and HeliPort connects to wifi, while SIP is still enabled. Thanks a lot for helping me sorting this out. I will keep testing this and report back if I see any crashes.

ResQre, 2024-09-24:
https://github.com/OpenIntelWireless/itlwm/issues/1009#issuecomment-2370893963

githubtohaoyangli, 2024-10-04:
macos15要求IO80211Family必须要求awdl没有就panic，实现awdl很费劲，目前缓解的方法就只能使用oclp1.6或mod版，但会破坏增量更新，且需部分关闭SIP。

githubtohaoyangli, 2024-10-04:
macos15要求IO80211Family必须要求awdl没有就恐慌，实现awdl很费劲，目前缓解的方法(原生WiFi，非itlwm+heliport)就只能使用oclp1.6或mod版，但会破坏增量更新，且需部分关闭SIP。

githubtohaoyangli, 2024-10-04:
如果实现了awdl会不会就能完整的airdrop，handoff等功能？