OpenIntelWireless/itlwm #984: Ax211 Ventura wifi working but in Sonoma not working

yzg011, 2024-06-13:
Environment

working well
Kext Version: AirportItlwm.kext.v2.3.0(Ventura )
WiFi Card Model:  ax211. (0x00948086)
macOS Version: Ventura 13.6

not working
Kext Version: AirportItlwm.kext.v2.3.0(Sonoma 14.4)
WiFi Card Model:  ax211. (0x00948086)
macOS Version: Sonoma 14.5

pls help me. thanks

yzg011, 2024-06-13:
all kext use the  stable version

yzg011, 2024-06-13:
wifi switch can't open bluetooth  is  working

zxystd, 2024-06-13:
attached the runtime log please, and can you post the hacklintool->pci info here? I don't see any intel wireless device that with device id 0x0094.

yzg011, 2024-06-13:
[image]
thanks for your quick answer
right now I am not in home

yzg011, 2024-06-13:
https://imacos.top/2024/05/11/14/

so many persons have this issue，but I try this way also， not working

zxystd, 2024-06-13:
4ac4c79bc7e34f8764038fc382630a29eb46213d
0x51F1 is already added in stable 2.3.0 version, are you sure you use the stable version downloaded in release page?

yzg011, 2024-06-13:
yes of course。 Sonoma 14.4 debug and stable version  have checked。 not working

andreszerocross, 2024-06-13:
There is default entry in config.plist for new opencore version. Make sure you're set it's "NO" or Disabled

[image]

yzg011, 2024-06-13:
I didn't set. any mini kernel value

yzg011, 2024-06-13:
> There is default entry in config.plist for new opencore version. Make sure you're set it's "NO" or Disabled [...]

will check and replay

yzg011, 2024-06-14:
> There is default entry in config.plist for new opencore version. Make sure you're set it's "NO" or Disabled [...]

solved ！thanks

closed 2024-06-14 by yzg011