OpenIntelWireless/itlwm #989: AX210 Wificard - Kext Version: 2.3.0 - Sonoma 14.5: Sometimes not work

Hafointosher, 2024-06-20:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 2.3.0
- WiFi Card Model: AX210
- PCI Product ID: 
- macOS Version: SONOMA 14.5 (23F79)

**Description**
In version 2.3.0 stable, my AX210 WiFi card sometimes doesn't work. Restart same issues. I tried version 2.3.0 beta and the WiFi works very well.

**Bug Report Archive**
I'm not using heliport

**Kext Download Source**
[GITHUB](https://github.com/OpenIntelWireless/itlwm/)

zxystd, 2024-06-24:
Which beta version? and please collect the runtime log if the stable version is not work.

C1014, 2024-07-06:
Yup, same problem. It is not consistent. I sometimes have to restart the wifi multiple times for it to work. The real problem is when you want to connect to a different wifi. I have to restart it multiple times for it to connect.

kamilkenrich, 2024-07-11:
I also encountered the same issue. The connection becomes stable after restarting the wireless router, but the problem reappears after a while.

Hafointosher, 2024-08-15:
I'm using itlwm v2.3.0 stable + Heli instead of AirportItlwm 2.3.0 stable. The Wi-Fi is working well. iService is not working, but I don't use it, so it doesn't matter. 👯