OpenIntelWireless/itlwm #990: AirportItlwm v2.3.0 stable Latest - Sonoma 14.5 iServices NOT working

rudo56, 2024-06-20:
> Support macOS Sonoma
>Users from macOS 14.0 to macOS 14.3 should use a separate kext due to an API change in 14.4+
>iMessage and related services will not work OOB with AirportItlwm.kext, use itlwm.kext or a builtin NIC as the primary device >instead

It is because 14.5?

itlwm + heliport make it work again

kimiazhu, 2024-06-20:
Same problem here on my NUC8i5beh，and my AirDrop works when I send files from Mac to iPhone, but failed in the opposite direction

rudo56, 2024-06-20:
Wifi cards are the big problem of hackintosh lately. Hope devs can give a hand.

pratiknborkar, 2024-06-21:
The same problem in my case as well I am using MacBookPro16,4 but my LAN port is not working so integrated EN0 from DSDT but still i-service stop working

rokipet, 2024-06-23:
it dosent work on sonoma 14.5 the same here i used vpn and it activates the iservices

rudo56, 2024-06-23:
> it dosent work on sonoma 14.5 the same here i used vpn and it activates the iservices

itlwm+heliport?
or Airportitlwm?

michaelliunsky, 2024-06-24:
> > it dosent work on sonoma 14.5 the same here i used vpn and it activates the iservices
> 
> 
> 
> itlwm+heliport?
> 
> or Airportitlwm?

itlwm+heliport worked well on 14.5 without vpn

rudo56, 2024-07-13:
> It seems that mac os verifies the connection via en0 - even if the cable is disconnected. With the cable disconnected, I shared the connection from WiFi to Ethernet (settings -> sharing) and tricked system into thinking the cable was connected (faulty Ethernet kext), assigned a static IP address for Ethernet, waited a while until the pings to Google stopped (now I have internet via Ethernet [useless but tricks system]) and deactivated Ethernet - iMessage messages were sent immediately. Let me remind you, I didn't have the cable connected even for a moment. The entire procedure must be repeated each time the system is restarted.
> 
> OS: Sonoma 14.5

Thanks for the answer. DO you think, this will be fixed in next releases?

350d, 2024-07-25:
> > > it dosent work on sonoma 14.5 the same here i used vpn and it activates the iservices
> > 
> > 
> > itlwm+heliport?
> > or Airportitlwm?
> 
> itlwm+heliport worked well on 14.5 without vpn

On my system `itlwm+heliport` introduce freezes and system slowdown... i9/64GB/6650TX

350d, 2024-07-25:
> It seems that mac os verifies the connection via en0 - even if the cable is disconnected. With the cable disconnected, I shared the connection from WiFi to Ethernet (settings -> sharing) and tricked [...]

I've managed to use integrated LAN with USB LAN adapter connected in loop with manual IPs. Wifi connection - first in a list. iServices works fine looks like. Sharing disabled.

MultimediaLucario, 2024-08-30:
Hopefully an update comes out soon for AirportItwlm for macOS Sonoma 14.5 that fixes the iservices bug. I am currently in college and I can't use their private Wi-Fi considering heliport does not support WPA2 Enterprise.

TommyLuco, 2024-09-09:
> Hopefully an update comes out soon for AirportItwlm for macOS Sonoma 14.5 that fixes the iservices bug. I am currently in college and I can't use their private Wi-Fi considering heliport does not support WPA2 Enterprise.

You could just use airportitlwm + nullethernet to create a "fake" ethernet adapter with a proxy (and then in priority make that nullethernet the lowest of your network adapters). that will fix your iservices

fidelio959, 2025-02-09:
> It seems that mac os verifies the connection via en0 - even if the cable is disconnected. With the cable disconnected, I shared the connection from WiFi to Ethernet (settings -> sharing) and tricked [...]

Hello can you explain better how to do this? I follow your guide but I always get the red bullet on ethernet connection. Cannot make it work.