OpenIntelWireless/itlwm #1000: Ice Lake-LP PCH CNVi WiFi

VuNex, 2024-07-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm.kext [2.3.0 sonoma 14.4+]
- WiFi Card Model: Ice Lake-LP PCH CNVi WiFi
- PCI Product ID: VEN_8086&DEV_34F0
- macOS Version: Sonoma 14.5

**Description**
Hi! The problem is that this device won’t start up. I’ve tried everything already. I also installed Sonoma 14.1, but nothing helped. I even tried modifying Device Properties. Can anyone help? I’ve been struggling with this for a week already.

[image]

**Kext Download Source**
https://github.com/OpenIntelWireless

zxystd, 2024-07-17:
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs
runtime log is needed.

VuNex, 2024-07-18:
I forgot to say that BT is working fine.

bilijp153, 2024-07-19:
I have the same hardware, but my WiFi is working fine.
[image]
[image]

VuNex, 2024-07-22:
> I have the same hardware, but my WiFi is working fine. ![截屏2024-07-19 21 34 36](https://private-user-images.githubusercontent.com/86920625/350412496-b1b4cabb-19ec-4e48-ad17-8fa31a22cd60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MjUzNTAsIm5iZiI6MTcyMTYyNTA1MCwicGF0aCI6Ii84NjkyMDYyNS8zNTA0MTI0OTYtYjFiNGNhYmItMTllYy00ZTQ4LWFkMTctOGZhMzFhMjJjZDYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA1MTA1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI5Mzc2OTE4ZGJkNjU4MTVjYTIwMmRlNTIyMmQ3OWVlYzMzOTQ0ZWFlMmE2MWE1Zjg3MTU3MTAyYmFjZTdlYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CmLSkzv-vmxRW6Om77FPW4kInjvhQ-cLn_qDGoL6mfA) ![截屏2024-07-19 21 34 20](https://private-user-images.githubusercontent.com/86920625/350412495-964e923c-2a1a-482f-bbf9-77717d086893.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MjUzNTAsIm5iZiI6MTcyMTYyNTA1MCwicGF0aCI6Ii84NjkyMDYyNS8zNTA0MTI0OTUtOTY0ZTkyM2MtMmExYS00ODJmLWJiZjktNzc3MTdkMDg2ODkzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA1MTA1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNiNzVlY2FhYTBkYzUyMjY4ZTBlYmE2ZTIzZGUwNjdiMjM1NmRjNTg5OWU0ZTQ1OWUxMjAyNmI2NzFjNzM3ZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kRWw3WTd4iCEW0b7YXlHscqPBHK3Aa9wv7aVFDw0EOI)

hat's strange... maybe my WiFi module is dead. The thing is, I bought a new computer without an OS and installed macOS on it... I'll check everything again and get back here.

VuNex, 2024-07-23:
> I have the same hardware, but my WiFi is working fine. ![截屏2024-07-19 21 34 [...]

what is the mac OS version?

bilijp153, 2024-07-24:
> what is the mac OS version?

macos 15 Sequoia Beta

wern-apfel, 2024-07-25:
No issues here with the same WIFI card. I have installed macOS on more than 10 laptops with an intel WIFI with this key and it has worked on all of them. Many thanks to zxystd for the great work.
I usually always deactivate HPET, it is no longer used by macOS since Skylake. Disabling HPET could fix your WIFI.

zxystd, 2024-09-21:
```
[    0.746586]: itlwm: Found device: Intel(R) Wi-Fi 6 AX101 fw: iwlwifi-Qu-c0-hr-b0-68.ucode index: 167:208
[    0.746590]: itlwm: Found PCI dev 34f0/0244, rev=0x338, rfid=0x10c000 fw: iwlwifi-Qu-c0-hr-b0-68.ucode
```
Seems that it is ax101？can you use Ubuntu-live to check if it is the same?

```
[    1.043571]: itlwm: iwx_cmd_done: command 0xc00 done
[    3.044681]: itlwm: iwx_stop_device
[    3.067987]: itlwm: iwx_check_rfkill RF_KILL hw: 0
```
the card not alive after initialization complete, not sure if it is hardware or firmware issue. but ax101 is confirmer working by other people.