OpenIntelWireless/itlwm #1001: OneXplayer1S AMD has BT but no Wi-Fi

WiziBox, 2024-07-17:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: 2.3.0
- WiFi Card Model: Intel 8265NGW, AC3160 
- PCI Product ID: 8086:24FD
- macOS Version: 13.6.7

**Description**
OneXplayer1S AMD has BT but no Wi-Fi. Airportitlwm is not working too. Windows does not have problem.

**Bug Report Archive**
[bugreport_11251.zip](https://github.com/user-attachments/files/16262793/bugreport_11251.zip)

**Kext Download Source**
GitHub Release

WiziBox, 2024-07-17:
[bugreport_35957.zip](https://github.com/user-attachments/files/16264682/bugreport_35957.zip)

Tried AX210(8086:2723), but still not working.

zxystd, 2024-08-14:
``
[    2.114907]: itlwm: : can't find MSI interrupt controller
``
PCIE or BIOS/ACPI error, cause no interrupt controller there, nothing can be do in this driver, contact the manufacture for help, thanks.

closed 2024-08-14 by zxystd