OpenIntelWireless/itlwm #1005: Airportitwlm 2.3.0 Stable Sonoma 14.4 not working on Sonoma 14.6.1

hartmut181059, 2024-08-24:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version:   2.3.0 Stable
- WiFi Card Model: Elitebook 840 G3 Internal 
- PCI Product ID: 
- macOS Version:  Sonoma 14.6.1

**Description**

Wifi under 14.6.1 not working, 14.4.1 is working

**Bug Report Archive**

**Kext Download Source**

Mauri13, 2024-08-25:
Kext Version: 2.3.0 Stable for Sonoma 14.4 work well on Sonoma 14.6.1 Intel CNVi interface 802,11a (z390)

hartmut181059, 2024-08-27:
Sorry I have done a new plain Installation and now it works with itlwm,kext and heliport Thx and excuse

arg0WAK, 2024-09-08:
It does not work properly in 14.6.1. DHCP is not assigned automatically. If entered manually, it connects to the WiFi network.