OpenIntelWireless/itlwm #1006: AirportItlwm AX210 Sonoma 14.5 Repetitive crashs

CommandMaker, 2024-08-27:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Last stable (2.3.0_stable_Sonoma14.4)
- WiFi Card Model: Wifi 6 AX210/AX211/AX41 160MHz (AX210)
- PCI Product ID: 0x8086 0x2725
- macOS Version:  Sonoma 14.5

**Description**
Hello !

Sometimes, while using AirportItlwm, the hackintosh crash and reboot without any apparent reason. Today, it crashed 2 times in a row just after started (no time to open the session) and sometimes, there isn't any crash for several days.
Occasionally, it crashs when using the macOS after several hours.

I don't know what can cause this (if this is a bad configuration, or a problem in the kext), I posted [here](https://pastebin.com/gfGVJA07) the crash report of the last one.

Thanks for your hard work and have a nice day !

**Bug Report Archive**

**Kext Download Source**
[Official release](https://github.com/OpenIntelWireless/itlwm/releases)

goof320, 2024-09-01:
i have also been having this issue! thank you for posting this, i'll share my crash report as well. i have the same wi-fi card (AX210), version of the kext and version of macOS as you. 
[log.txt](https://github.com/user-attachments/files/16826794/log.txt)

madvinegar, 2024-09-16:
I have the same card and same problems. I find it to be stable at 2.4ghz. (it crashes when connected to 5ghz)

zxystd, 2024-09-21:
[Sonoma14.4.zip](https://github.com/user-attachments/files/17086058/Sonoma14.4.zip)
please try this one and see if it can help.

CommandMaker, 2024-09-22:
Hello,
Sorry but it didn't worked, it just always reboot just after the libignition without any kernel panic (even with kernel panics show set to true).
I put a [video](https://drive.google.com/file/d/1RVxjHZM9Tg0_y3jTM048wkvIN0Fr1xux/view?usp=sharing) to show the boot logs if it can help

zxystd, 2024-09-22:
Configuration is correct? panic logs should be generated if it is caused by this kext.