OpenIntelWireless/itlwm #1008: Itlwm + Heliport Loop '!' Symbol on Recognised Network

Sooly890, 2024-09-05:
**Have You Read Our Docs**

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: Newest
- WiFi Card Model: Intel Dual Band Wireless AC 3168
- PCI Product ID: 0x24FB (This was the Device ID in Hackintool, please tell me if Device ID and Product ID are not the same)
- macOS Version: 14.5

**Description**

When I boot my computer then log into my second account (the account I made after the installation), itlwm seems to look for networks, connects with the `!` symbol, then disconnects, and then repeats. It also happens on my first account (the one I created when installing), when I sign out of my second account then sign into my first account. This does __not__ occur when:
- I sign into my first account
- I sign into my first account, then logout, then sign into my second account.

**Bug Report Archive**

[bugreport_45452.zip](https://github.com/user-attachments/files/16884278/bugreport_45452.zip)

**Kext Download Source**

Got itlwm from newest Release on Github, got Heliport from newest Release on Github.
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.3.0
https://github.com/OpenIntelWireless/HeliPort/releases/tag/v1.5.0

zxystd, 2024-09-20:
password incorrect, maybe you should log in the 'wrong' account and then forget password in the Heliport, then restart.

Sooly890, 2024-09-21:
Thanks for responding [zxystd](https://github.com/zxystd), I'll try that now

Sooly890, 2024-09-21:
Thank you very much [zxystd](https://github.com/zxystd), it worked like a charm

closed 2024-09-21 by Sooly890