OpenIntelWireless/itlwm #1009: Please add AirportItlwm support for macOS 15 Sequoia

liangcepaul, 2024-09-19:
Please add AirportItlwm support for macOS 15 Sequoia

SudoFormatAll, 2024-09-19:
+1 vote

tinodin, 2024-09-19:
+1 vote

wolflonezx11, 2024-09-19:
Please add AirportItlwm support for macOS 15 Sequoia
AirportItlwm support for macOS 15 Sequoia PleasePleasePleasePlease

rifat151, 2024-09-20:
AirportItlwm support for macOS 15 Sequoia

liangcepaul, 2024-09-20:
When was it supported?

liangcepaul, 2024-09-20:
> AirportItlwm support for macOS 15 Sequoia
> When was it supported?

fareedkh, 2024-09-20:
+ Vote
Please add AirportItlwm support for macOS 15 Sequoia

lopess1971, 2024-09-20:
Please add AirportItlwm support for macOS 15 Sequoia

jzrodriguez98, 2024-09-20:
Please add macOS 15 support to Airportitlwm

liangcepaul, 2024-09-20:
@zxystd

naidb, 2024-09-21:
Please add macOS 15 support to Airportitlwm +1

Zormeister, 2024-09-21:
+1ing won't change anything as of current tbh, but i have faith that zxystd can do it

give him time, IO80211Family is massive.

royalgraphx, 2024-09-21:
ETA kids be like:

Serghei24Y, 2024-09-21:
Hi, Developer create AirportItlwm for Sequoia. My Intel WIFI 6E module worked fine on Sonoma 14.5 and below. It will be great if AirportItlwm works on macOS 15 !!!

Blu-Dread, 2024-09-21:
Hi I try to build the project but I am stuck .
[image]

lifeknife10A, 2024-09-21:
A bit off topic does your Bluetooth work because mine doesn't. Intel AC Wireless 9650

Blu-Dread, 2024-09-21:
mine doesn't work because it's on the same wifi card

royalgraphx, 2024-09-21:
> Hi I try to build the project but I am stuck . ![Schermopname 21-09- ](https://private-user-images.githubusercontent.com/124017731/369618496-088ff0ed-1c40-4f7e-9357-8577b857b590.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5MjMwOTgsIm5iZiI6MTcyNjkyMjc5OCwicGF0aCI6Ii8xMjQwMTc3MzEvMzY5NjE4NDk2LTA4OGZmMGVkLTFjNDAtNGY3ZS05MzU3LTg1NzdiODU3YjU5MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQxMjQ2MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZTljZDllNzk2MDEzNGM1MWIwMGZjNDc0MzlmNDc5ZjljMTNiM2I0Mjg0MzgyYzEwOWZlOTk3MDQ2ZjY3Nzk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sZekPL5KOxWuz2N5XdZsfJfYdOeYAi02M4CSsMq-Alc) because of some permission reason .

This is a GitHub issue, not a place for discussion. Please read your logs and look at the very last line, verify the shell script has execution rights.

For others, there is nothing you will say in this Issue that will make it come any faster, please, act like you have used Github before.

SudoFormatAll, 2024-09-21:
> +1ing won't change anything as of current tbh, but i have faith that zxystd can do it

well, you are not right at least because you should not "affirm" to the public that you do not know for sure, because you are not the author and did not receive an answer from him, then these are just your fantasies.
In the votes there is at least one meaning that the author can calculate "necessity" for his work and manage time more neatly depending on the number of users.
For example, if there are only 2-3 people, he **can** decide not to waste his time and stop the development, or will only do it at a time when he is completely bored ...
so, **he** decide, not **you** if this information can be useful.

royalgraphx, 2024-09-21:
@SudoFormatAll , do you even know who Zormeister is? He actually as a matter of fact has had contact and was even attempting to help on this front. On top of that, do you _really_ seriously think, that people adding a +1 in this issue is going to raise the priority for zxystd? He is **obviously**, and painfully aware, that macOS 15 Sequoia does NOT have support, because of the massive API changes in said release. Do you think he thinks it just works on macOS 15? How dense do you have to be, to feel like you need to remind a developer, what does and does not work?

with a little bit of common sense, you'd know, that this thread literally does nothing for him, or any developer. People begging and being entitled to an update, does NOT make it come any faster, or makes his reverse engineering ANY easier.

SudoFormatAll, 2024-09-21:
> How dense do you have to be

Your post is too rude to a person who does not know how to read, so I will answer with your words: "How dense do you have to be to can not be able to read?"
This topic does not "remind a developer, what does and does not work", you are an moron, I wrote that it only describes the number of people, who need his job. so this only strengthens my words about the meaning of votes, because the more complicated the work, the more meaning is needed for a waste of time on it, and if your work is not needed, then .. in general you do not understand at all that I wrote.
What about the "Zormeister" , i dont know every user on network, but the fact that he "tried" does something in front not make him the author, or super-star, so do not try to come up with what it makes no sense with my words. Or is this your second account? However, this is not important, because you are talking about completely different things....

royalgraphx, 2024-09-21:
> Your post is too rude to a person who does not know how to read

I have 0 obligation to be nice.

> This topic does not "remind a developer, what does and does not work", you are an moron, I wrote that it only describes the number of people, who need his job

you literally just said the same exact thing, in a different way. If you cannot realize that, then you are the moron.

> because the more complicated the work, the more meaning is needed for a waste of time on it

Tell me you've never been a developer before, without telling me. Do you not realize that people stacking on top of an issue that is already being worked on behind the scenes ever since macOS 15 went into Developer Beta, is suddenly going to ramp up in production because a bunch of end users in a _Github Issues_, ended up turning it into a discussion about how badly they need an update? do you lack critical thinking? or are you simply that entitled that you really truly believe, you're going to make it known to zxystd he basically needs to ``hurry up``, because there's more people waiting on him. Do you not realize this does the complete opposite? by actually stressing a developer out?

> in general you do not understand at all that I wrote.

No, you are incredibly entitled, and think that somehow, spamming a github issue, is going to make zxystd hurry up, when he's literally already been putting in some effort, even if its minimal, but you lack the understanding that, the amount of effort he puts into this, is NOT equal, to the amount of people that _complain_

> i dont know every user on network, but the fact that he "tried" does something in front not make him the author, or super-star, so do not try to come up with what it makes no sense with my words.

First of all, I can now CLEARLY see the language barrier because this sentence is an absolute mess. First of all, by informing you that the person you so easily said was a nobody, and has never had contact with zxystd, was an attempt for you to grasp a bit of understanding as to why he even made the comment in the first place. "tried" does not mean he failed. You are assuming he failed because of your poor english, "try" in this context, means he lended his help to zxystd to make this update come faster.

SudoFormatAll, 2024-09-21:
> you literally just said the same exact thing, in a different way. If you cannot realize that, then you are the moron.

wow i see a mentally ill person in the thread...
everything i wanted to say in my answer "Zormeister" is not connected with ALL crazy nonsense that you wrote...and you only one who have language barrier with it. 
i just said that "the **author** decides what is useful and what is not" and not a random user. and the fact that Zormeister is not even in [Contributors] only confirms the fact that he is not even a representative of the author. 
**so my message only about that you don't need to pretend to be an important person who represents the interests of the author.**
you can be a developer for a thousand years and understand his feelings, or anyone else and even have one or two conversations with the author, but neither you and he are not author, not his representative. this ALL i want to said.
is it so hard to understand? if so then you need to treat a sense of self -importance.

royalgraphx, 2024-09-21:
> you can be a developer for a thousand years and understand his feelings, or anyone else and even have one or two conversations with the author, but neither you and he are not author, not his representative. this ALL i want to said.

So basically, after everything, you still do not understand. You simply do not get it. Clearly, there is no intellectual conversation to be had here, you're an entitled end-user who thinks he'll get an update faster in this thread, and have no regard for your conduct when it comes to pestering a person who CLEARLY knows macOS 15 exists. You nor anyone, had to remind him of this fact. Again, his effort, is not equal to the amount of complaints or votes. Things are done on their own time.

As for not being on the contributors list or keeping things in DMs, how do you know there aren't unpushed commits? Not saying there are, but you keep assuming you have a stance because you can dismiss a user you have no idea about or what they do or are capable of, or what they even have done.

> but neither you and he are not author, not his representative

I did not realize common sense, and proper conduct of yourself in a public space, is tied to being directly an official representative. You can be surrounded by 50 people who don't know shit about rocket science, but if a guy tells you to stop bugging or interrupting the teacher because they're in the middle of a lesson trying to listen and learn, you don't tell him/her he's/she's not qualified to tell you to hush, just because he's/she's not an official representative or the teacher himself/herself.

SudoFormatAll, 2024-09-21:
> So basically, after everything, you still do not understand. You simply do not get it.

no, you don't understand, that's why I think you need to treat your sense of self-importance.
as I already said, you can be anyone, even the president of the galaxy, it has nothing to do with what I said. your abilities also don't give you the right to represent the author's rights. you can work in apple or save the planet, it has nothing to do with the fact that you decide for another person. you can understand his feelings, that's your right, but you definitely don't have the right to decide for others.
I'll finish here, because you clearly have an inflated self-esteem and you are too stupid to understand the main point of the claim, you only speculate on secondary things. There is nothing pleasant about continuing a conversation with people like you. bye.

royalgraphx, 2024-09-21:
>  it has nothing to do with the fact that you decide for another person

the only person i'm literally deciding for, are all the stupid people in this thread thinking a stupid ass ``+1`` is going to suddenly make zxystd push an update in 2 days, like how dumb do you have to be to really think thats how this works?

I'm not deciding for the author, i'm explicitly explaining to you, and everyone else in this thread who reads this, that their efforts are worthless. The update will come, when it comes. The amount of work or effort, is not equal to the amount of complaints or votes. Things are done on their own time. How many times do I need to repeat that, until you finally understand that the only point i'm attacking, is directly yours, where you say IT DOES make a difference, where I am clearly saying it does NOT.

You can end the conversation whenever you please, and you can keep assuming what you'd like, but I, am working on cold hard facts. Not emotion, nor opinion, I, KNOW, for a fact, that these +1's are pointless. Zorm is much nicer than I am, they said it as nicely as they could, I on the other hand, have 0 obligation to be nice. It is not self entitlement just because i'm being rude to someone who's dumb enough to really think they're going to make a difference by comming in an issue title ``Please add AirportItlwm support for macOS 15 Sequoia`` - 2 Days ago, when macOS 15 has been in developer beta and has been under reverse engineering for months (Jun 11, 2024). Since it was literally possible to download and install.

SudoFormatAll, 2024-09-21:
> I'm not deciding for the author
This is exactly what you do ... give your opinion for official facts .. idiocy.
> where I am clearly saying it does NOT.
As I already said, nobody care about what the patient and Rude noname User "clearly" said, even if he is captain America. Drink pills, calm down and when your "BIGG I AM" is cool down, then you will understand that your opinion is just your opinion, not official representation of anybody.

> you say IT DOES make a difference, where I am clearly saying it does NOT.
more than half of what you wrote, is **your** words, NOT mine, you speculate and fantasy up everything. I just wrote that it is “**can** calculate" "**can** decide”, **he** can, not **you**, I wrote the right words, I allocated them with bold font, no more than this. I never wrote anywhere that this would _accelerate_ the development, I never wrote that it would save the planet ... I just wrote that the author decides that it is useful, how to use it, will he use it or not... you just sick speculate troll.

royalgraphx, 2024-09-21:
LMAOO, this dude is a brick wall, enjoy waiting!

liangcepaul, 2024-09-21:
> A bit off topic does your Bluetooth work because mine doesn't. Intel AC Wireless 9650
@lifeknife10A It works fine on my laptop, Intel Dual Band Wireless-AC 3165

sughero, 2024-09-22:
I got everything working (that includes Audio and BlueTooth) BUT wifi.
[Here](https://github.com/nakquada/Z790-Hackintosh) is said that is a FULLY working configuration for MacOS Sequioia (included wifi).
I asked him which kext is using. I tried his kext but it doesn't work for me.

jzrodriguez98, 2024-09-22:
I was checking the repository and after that reviewed the motherboard specs in Gigabyte's website. The mobo has the following: 
Wireless Communication module
Intel® Wi-Fi 6E AX211
WIFI a, b, g, n, ac, ax, supporting 2.4/5/6 GHz carrier frequency bands
BLUETOOTH 5.3
Support for 11ax 160MHz wireless standard and up to 2.4 Gbps data rate

liangcepaul, 2024-09-23:
> I got everything working (that includes Audio and BlueTooth) BUT wifi.
> [Here](https://github.com/nakquada/Z790-Hackintosh) is said that is a FULLY working configuration for MacOS Sequioia (included wifi).
> I asked him which kext is using. I tried his kext but it doesn't work for me.

Me, too. I got everything working (that includes Audio and BlueTooth) BUT wifi doesn't work.

Matgen84, 2024-09-23:
@zxystd  Please add macos 15 Sequoia support to AirportItwlm

speedie1337, 2024-09-23:
To those who keep commenting 'Please update it for macOS 15': You are not paying the developer to work on this project. You have zero programming experience. Shut up. As @royalgraphx said, the update will be out when it is complete and if you're not happy with that you can go write it yourself.

royalgraphx, 2024-09-23:
if you guys want it to come any sooner, either contribute something yourself, or find us, me or Zorm, a way to directly contact zxystd in a way where we can actually have real-time communication instead of emails, Glitter, or Discord (These methods all seem dead, and has not yielded a response in a very long time, if at all on specific platforms). So that we can actually help. 

I understand that people are unaware of us (Carnations Botánica), we have more private repos than we do public, and the only public repo we did have, we had to make private, because people were finding out and joining prematurely. We haven't released any projects for end-users, but to dismiss a developer who actually has potential to help everyone here be satisfied with a sooner release, just because you don't know them, on top of being someone who has provided even more than just contributions to this Wi-Fi issue (LegacyRed, NootedRed, and other private work), is a horrible take. 

You guys just don't get it, **and that's fine**. Whether or not you have experience using GitHub, always conduct yourself like you are in a public space and do not know what others do. I have, and had 0 obligations to be nice, because I'm coming from a position where I can speak on what the real issue is (API changes that require RE, that cannot be done alone as it is incredibly time consuming), and as i've stated so many times:

``Do you guys seriously think, zxystd is unaware that macOS 15 exists?``

Please, use any brain cells left, to genuinely understand that is a stupid thought to have, in any capacity, he is aware. No one has to remind him, and no voting will make a difference. Priority does not come from the amount of people "voting" which is truly just simply complaining and stressful, to a developer who is 9/10 literally taking a fucking break.

I would myself, and equally encourage ANY team members to do the same, if this is the community we had over at Carnations Botánica as things release OR fall behind in updates. Most of the time, developers do not do this for _you_, they realize they themselves need something, and willingly, and NICELY, i may add, release it to the public. **For Free**.

ResQre, 2024-09-24:
Give me some time, I will write a tutorial about the OCLP method to make intel Wi-Fi work on Sequoia.

ResQre, 2024-09-24:
The tutorial of OCLP method to active intel Wi-Fi card on Sequoia, includes two parts. One is OCLP Wi-Fi root patch and the other one is faking intel Wi-Fi card as BCM one to make OCLP root patch working. The details are listed below.

Use Hackintool to copy your device path of Wi-Fi card.
[image]

Edite your config, add the device path above without the "#" and fill all the stuff like mine. This step is faking intel Wi-Fi as BCM one, then OCLP root patch can be applied.
[image]

Go to OCLP [website](https://github.com/dortania/OpenCore-Legacy-Patcher/tree/main/payloads/Kexts/Wifi), download the kexts, in order like mine.
[image]

Block the kext.
[image]

Load "ventura"-AirportItlwm.kext.
After reboot, then you can do the OCLP root patch.
[image]

Reboot and put "#" in front of the device path to disable the fake ID. Then reboot. Done!

This is a temporary fix to make AirportItlwm.kext woking on Seqouia. Airplay and iServices are fully working. AirDrop is oneway working. I'm still waiting for the Sequoia-AirportItlwm.kext. 

Thanks for your great work! @zxystd

lifeknife10A, 2024-09-24:
> The tutorial of OCLP method to active intel Wi-Fi card on Sequoia, includes two parts. One is OCLP Wi-Fi root patch and the other one is faking intel Wi-Fi card as BCM one to make OCLP root patch working. The details are listed below.
> 
> 
> 
> Use Hackintool to copy your device path of Wi-Fi card.
> 
> [image]
> 
> 
> 
> Edite your config, add the device path above without the "#" and fill all the stuff like mine. This step is faking intel Wi-Fi as BCM one, then OCLP root patch can be applied.
> 
> [image]
> 
> 
> 
> Go to OCLP [website](https://github.com/dortania/OpenCore-Legacy-Patcher/tree/main/payloads/Kexts/Wifi), download the kexts, in order like mine.
> 
> [image]
> 
> 
> 
> Block the kext and set csr-active-config 03080000.
> 
> [image]
> 
> 
> 
> Load "ventura"-AirportItlwm.kext.
> 
> After reboot, then you can do the OCLP root patch.
> 
> [image]
> 
> 
> 
> Reboot and put "#" in front of the device path to disable the fake ID. Then reboot. Done!
> 
> 
> 
> This is a temporary fix to make AirportItlwm.kext woking on Seqouia. Airplay and iServices are fully working. AirDrop is oneway working. I'm still waiting for the Sequoia-AirportItlwm.kext. 
> 
> 
> 
> Thanks for your great work! @zxystd 

Thank you @ResQre .

I will definitely Let you know whether this solution worked for me. Also if i get stuck anywhere i will definitely require your guidance.

Again, Thank you so much 🙏🏻🙏🏻

royalgraphx, 2024-09-24:
While I think the above guide is an okay approach for people who are very desperate, please allow my two cents.

1. OCLP is not supported on Hacs. They are very very adamant on this fact.
2. Please, stop using configurators. There are so many technical reasons, but let me give you an example of why using anything but [ProperTree](https://github.com/corpnewt/ProperTree) is a bad idea.

What you all know as a .plist file, is simply a specially crafted XML file by Apple, that is used for various information on the system, it is a properly list file, it contains properties in a list. Now, that's all well and good, but did you know that configurators actually have no clue how to properly parse and manage .plist files? It's true. ProperTree, is known as a cross-platform GUI plist editor written using Python. What does this mean for you? If you've ever used a Mac before, and have ever used Xcode, you would know that plist files are editable via a special Xcode Plist Editor of sorts. Now, even this, is not suitable for the kind of plist format that OpenCore uses/expects. There is an amazing example of this:

If you have configured SecureBoot, and ApECID in your config.plist, and you open this plist in a Configurator two things will happen: It will attempt to parse the plists various data, and because they're not understanding of specific Data, will literally ``0`` your ApECID. After it does this for every other invalid data, it will then **automatically** save the plist. This happens every time you open your config.plist with a configurator, you are risking the application incorrectly understanding your data, and will zero it out, WITHOUT telling you. No popup, no warning, nothing, you will now have, an non-bootable system.

For the love of everything, please, stop using configurators, and use the literal python rewrite, of the official Xcode PLIST Editor, that is specially designed and focused to respect OpenCore's expectations. I understand it may be hard to use ProperTree, but... really? It's quite literally, just a plist editor and **_thats the point_**.

royalgraphx, 2024-09-24:
also, it's incredibly clear now, that the users in this thread are incredibly unwilling to learn anything, or are incapable of doing so. This will be my last message, I will now let you guys rot.

ResQre, 2024-09-24:
@royalgraphx Sorry to make you feel discomfort. I agree with your consent and will drop my post.

royalgraphx, 2024-09-24:
@ResQre I cannot stop you, nor would I feel good about you taking your post down, my entire point is _teaching_ people. I don't want people flooding the OCLP discord, they do not help with Hacs, you, are not on a macbook air, you can go and buy supported wifi and swap yours out, you, can use and learn how to use ProperTree. It's not hard, my entire goal, is to simply educate people. There are right ways to do things, and there methods, that will result in bad outcomes, that is all

b00t0x, 2024-09-24:
While it's true that OCLP isn't for Hackintosh, it's clear from [this notes](https://github.com/dortania/OpenCore-Legacy-Patcher/pull/1077#issuecomment-1646934494) that they are not "very very adamant" about root patches being used on Hackintosh. Of course we shouldn't bother the OCLP Team with Hackintosh.

OCAT is an EFI manager based on the plist editor xplist and is different from OpenCore Configurator, but this is not a topic to discuss in this issue.

@ResQre Good tutorial, but there are a few notes:

1. Only `IOName` is needed in DeviceProperties
2. AirPortBrcmNIC.kext is not needed
3. MinKernel should be 24.0.0, as this hack is only needed for Sequoia

royalgraphx, 2024-09-24:
You are referencing a very old PR, that has the following notes:

``While the project is designed for legacy Mac hardware, we know the community is quite interested in our development of Broadcom patches. For those who wish to use the Broadcom patches on non-standard machines such as Hackintoshes, see below``

This, is not a co-sign from MykolaG to use OCLP on Hacs, he is simply acknowleding the fact that no matter what he or Dhinak do, people will NOT stop using OCLP on Hacs, because people cannot get that through their head. So no, this is not any form of allowance, and simply a matter of defeat, and understanding nothing they can do, will ever stop people from bothering them.

OCAT, OCC, and various others, are all configurators. When I explicitly state, to stop using Configurators, I am talking about **all of them**, they all have the ability to _configure_ a plist. Not just that -- they are all destructive in nature, when it comes to properly parsing a PLIST, it does NOT matter what library or whatever its based on, CorpNewt has wasted years refinding an exact copy of the official Xcode plist editor, that is SPECIFICALLY designed, to interact with plists that follow a specific structure, the one that OpenCore itself expects. CorpNewt is not a random person, that is a person who has been in the community and has served over 70,000 machines in his various servers and servers he interacts with. If you don't take that as proof of someone creating a tool that is literally meant for OpenCore, wtf else can I do? Nothing I say, will ever convince you otherwise.

sughero, 2024-09-24:
I am getting crazy! I followed @ResQre tutorial but I am still missing something I guess!
When I reboot OCLP says there's nothing to patch! (That menu is greyed out)
[image] 
This is what I got from Hackintool
[image]
Is there something wrong here?
[image]

jzrodriguez98, 2024-09-25:
I prefer to wait until the developer adds macOS 15 support to the airportitlwm kext instead of going through the hassle of applying root patches which was a method I decided to avoid to get my Fenvi card working in Sonoma. I got an Intel wifi and will continue to use itlwm + heliport which is good enough. As for the tutorial above, I think faking the card in device properties requires more than what you did- take a look again at the screenshot included in the tutorial.

goodbest, 2024-09-25:
@sughero If you have any issues in faking intel wifi card's device properties, you can instead try [this](https://github.com/goodbest/HaC-Mini/releases/download/v2.9.1/OpenCore-Patcher166.pkg.zip) very old build of OCLP ( a nightly build v1.6 in July, 2024).

The root patching option is not grayed out.

rbelusko, 2024-09-26:
> > The tutorial of OCLP method to active intel Wi-Fi card on Sequoia, includes two parts. One is OCLP Wi-Fi root patch and the other one is faking intel Wi-Fi card as BCM one to make OCLP root patch working. The details are listed below.
> > Use Hackintool to copy your device path of Wi-Fi card.
> > ![1](https://private-user-images.githubusercontent.com/75792086/370241061-357799cc-3bf7-4347-932d-ae73536db03a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTQ4MTMsIm5iZiI6MTcyNzM1NDUxMywicGF0aCI6Ii83NTc5MjA4Ni8zNzAyNDEwNjEtMzU3Nzk5Y2MtM2JmNy00MzQ3LTkzMmQtYWU3MzUzNmRiMDNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI2VDEyNDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllNmY5YjViOGZmZDAyNDYwZTc1YWMwYmNmMDM3MTY5NzFmMDIyZTYzZjAzODc0MzQ2NzdjN2VjNDc3YmViODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7LGcj-biX0hEAF5gzhQ6bN51VkqB87jvKJgtvJQtyow)
> > Edite your config, add the device path above without the "#" and fill all the stuff like mine. This step is faking intel Wi-Fi as BCM one, then OCLP root patch can be applied.
> > ![2](https://private-user-images.githubusercontent.com/75792086/370241249-c18aecfa-dbb1-425b-9bcc-4c9dbcfc44ca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTQ4MTMsIm5iZiI6MTcyNzM1NDUxMywicGF0aCI6Ii83NTc5MjA4Ni8zNzAyNDEyNDktYzE4YWVjZmEtZGJiMS00MjViLTliY2MtNGM5ZGJjZmM0NGNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI2VDEyNDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiZDljYmM0NzQyMTllOWQxZmM3ZWRkMjlmNTIwYTg2MzYwMTIxN2VmMDJhYTQzNTNmOTM0Yzk4MmE1MzRjOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.40-vrwieo0EkjbrXsM9XWj2OnGRladlYWkes5PcdmQ8)
> > Go to OCLP [website](https://github.com/dortania/OpenCore-Legacy-Patcher/tree/main/payloads/Kexts/Wifi), download the kexts, in order like mine.
> > ![3](https://private-user-images.githubusercontent.com/75792086/370241357-c5fdb6bb-a230-4ff0-abbd-581375a2f766.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTQ4MTMsIm5iZiI6MTcyNzM1NDUxMywicGF0aCI6Ii83NTc5MjA4Ni8zNzAyNDEzNTctYzVmZGI2YmItYTIzMC00ZmYwLWFiYmQtNTgxMzc1YTJmNzY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI2VDEyNDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA3YmExNTcwNzUxMGJhZTI0ODc3Zjg2NjRjNjY0ZTEwYjUxNGY1YjliNTBlNjJiMTg4ODJjYjQ1N2QxY2NkOTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6WZ3PzqpO51O72EN19w6naRK4f5RHWcQ4bVFHg3odoY)
> > Block the kext and set csr-active-config 03080000.
> > ![4](https://private-user-images.githubusercontent.com/75792086/370241449-c0b0d4b2-77fb-40b5-bef9-18b069148940.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTQ4MTMsIm5iZiI6MTcyNzM1NDUxMywicGF0aCI6Ii83NTc5MjA4Ni8zNzAyNDE0NDktYzBiMGQ0YjItNzdmYi00MGI1LWJlZjktMThiMDY5MTQ4OTQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI2VDEyNDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkZDg3MjVlZGYwMzM2Y2MxMGJiY2M4NmRhOWRlZWI2ODVmOTdhODY2NDA2MGQ1ZWJhNDJiNDM1NGVjOTU0ZTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ziHw3B0TAJh0rd-CnG1QKomxATFkcS65bEdXAiVPq0Y)
> > Load "ventura"-AirportItlwm.kext.
> > After reboot, then you can do the OCLP root patch.
> > ![截圖 2024-09-24 18 19 27](https://private-user-images.githubusercontent.com/75792086/370241865-e359a93c-7968-42e6-b713-6934c6b5678b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTQ4MTMsIm5iZiI6MTcyNzM1NDUxMywicGF0aCI6Ii83NTc5MjA4Ni8zNzAyNDE4NjUtZTM1OWE5M2MtNzk2OC00MmU2LWI3MTMtNjkzNGM2YjU2NzhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI2VDEyNDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4ZTNhMThkNTQwOTc3ZDAyODUwMjgzNzU2NzQ1Yjk3MjNhNDBhM2JkOThiMmFlZjI4Yzg3OTc0OTA1MGQyNTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PUYvBqjNy65kmdzOEaSSo_D_JYmdTL9Atg3x-a2pLEo)
> > Reboot and put "#" in front of the device path to disable the fake ID. Then reboot. Done!
> > This is a temporary fix to make AirportItlwm.kext woking on Seqouia. Airplay and iServices are fully working. AirDrop is oneway working. I'm still waiting for the Sequoia-AirportItlwm.kext.
> > Thanks for your great work! @zxystd
> 
> Thank you @ResQre .
> 
> I will definitely Let you know whether this solution worked for me. Also if i get stuck anywhere i will definitely require your guidance.
> 
> Again, Thank you so much 🙏🏻🙏🏻

I followed all the steps but was unable to get Native WiFi working.
I have an Intel AX 211 card on my Z790 MoBo. 12900KF with Radeon 6600.

[... cut ...] h points at the Intel WiFi chip)?

I did everything your guide stated:
1. Faked Broadcom using the device path ID of my WiFI, with the properties you provided.
2. Added the IOSkywalk downgrade values (kexts, block and NVRAM edits), applied Airportitlwm kext.
4. Rebooted, applied OCLP patches
5. Removed Fake Broadcom from device properties (put # in front of entry).
6. Rebooted, but no native WiFi active.

What did I miss? I like Native WiFi for FindMy services (doesn't work with ITLWM+Heliport).

ResQre, 2024-09-26:
@sughero Yes, the contents in your last picture with black circle mark are totally wrong. You should fill in the same contents as mine not your intel one, because we want to fake it as a BCM card. 
@rbelusko Since you made OCLP root patch working, that means you are doing right. All you have to do is to check your kext's version, esp. "Ventura"-Airportitlwm.kext.

This OCLP method restore Ventura-Airportitlwm.kext which means the frustrated iServices fail on Sonoma can be fixed,too.

rbelusko, 2024-09-26:
Followed the same set of steps on my 2nd hackintosh (Dell Latitude 5511). Still not able to active the native WiFi.
Definitely using Airportitlwm for Ventura (2.3.0).
I can post my OC config.plist... but i followed exactly what you wrote, on both machines.

ResQre, 2024-09-26:
AMFIPass
IO80211FamilyLegacy
IOSkywalkFamily

rbelusko, 2024-09-26:
I use "amfi_get_out_of_my_way=1" in NVRAM. Isn't that the same thing?
Also I had IOSkywalkFamily above IOFamilyLegacy. Does that order matter?

rbelusko, 2024-09-26:
I don't have access to AMFIPass 1.4.1. The one I've found available was on a 3rd party github page, I can't be sure it's 1.4.1.

ResQre, 2024-09-26:
All the kexts are available in the link I posted. 
If you still fail, add -lilubetaall.

rbelusko, 2024-09-26:
Getting a little off topic so I apologize... so the AMFIpass1.4.1 Kext is different than using NVRAM edits?
I was able to find the kext in your link, I will test in a few hours and let everyone know.

ResQre, 2024-09-27:
[image]

ResQre, 2024-09-27:
[image]

ResQre, 2024-09-27:
[image]

LGhostWhat, 2024-09-27:
Hi! I was trying for too long and encountered this post. I followed as much as I could and couldn't bring myself to enable native Wi-Fi. Then I supposed it could be related to the order of the kexts as @ResQre said so I put Ventura's AirportItlwm on the bottom of the chain and I managed to enable it. Thanks!

ResQre, 2024-09-27:
My mistake. AirportItlwm needs IOSkywalkFamily and IO80211FamilyLegacy loading first to hook up.

[132 more replies]