OpenIntelWireless/itlwm #1011: Please add MacOS sonoma 14.5 support

Dellguy-ux, 2024-09-24:
I installed macOS 14.5 on an offline installer and there is no airportItwlm kext for macOS 14.5 if you can compile it thanks!

jzrodriguez98, 2024-09-24:
Release 2.3.0 provides support to Sonoma 14.5? Have you tried the airportitlwm kext included in this release? I used that version while I used 14.5.

hlbomfim, 2024-09-24:
sequoia, please  :'(

Dellguy-ux, 2024-09-25:
> I had no issues upgrading from Sonoma 14.4 to 14.5. As @jzrodriguez98 said the [Airportltwm for Sonoma14.4](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0/AirportItlwm_v2.3.0_stable_Sonoma14.4.kext.zip) should still work after the upgrade to 14.5 (not to 15 though!)

but I installed macos sonoma not upgraded I didn't upgrade from macos 14.4 to 14.5

Dellguy-ux, 2024-09-25:
thank you it works now!

closed 2024-09-25 by zxystd