OpenIntelWireless/itlwm #1013: AX200 constant random disconnects

elfametesar, 2024-09-26:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**
Sonoma
- Kext Version: Airportitlwm v2.3.0
- WiFi Card Model: AX200
- PCI Product ID: idk
- macOS Version: 14.5

**Description**
Sometimes just the internet stops working for which i have to reconnect to the wifi and sometimes the wifi disconnects. I hope the log contains the issue, if not, i could provide another. I know there have been several issues opened for this but I couldn't find a solution in them

**Bug Report Archive**
[itlwm_logs.log](https://github.com/user-attachments/files/17154617/itlwm_logs.log)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0/AirportItlwm_v2.3.0_stable_Sonoma14.4.kext.zip

elfametesar, 2024-09-27:
Similar issue with itlwm + heliport, didn't notice wifi drops but internet stops working constantly and have to reconnect to wifi

[itlwm_logs.log](https://github.com/user-attachments/files/17165431/itlwm_logs.log)

hassy-ml, 2025-09-04:
getting the same issue