OpenIntelWireless/itlwm #1015: Itlwm is not running but kext is loaded

emanuele-pogliari, 2024-09-27:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: itlwm 2.3.0 stable
- WiFi Card Model:  Intel AX210
- PCI Product ID: 0x2725 (maybe?)
- macOS Version:  Sonoma 14.7

**Description**
kext results loaded but heliport says itlwm is not running
[image]

from hackintool. Card is recognized but not on the system
[image] 

my pc is an amd mini pc (beelink SER 5 PRO). 

I tried airportwlm, tried to force PCI Device Path in config.plist, remapped usbs, tried everything...

**Bug Report Archive**
[bugreport_23529.zip](https://github.com/user-attachments/files/17164161/bugreport_23529.zip)

**Kext Download Source**
just used kext from this github page.

Takudzwamz, 2024-10-01:
I'm facing the same issue as well

Takudzwamz, 2024-10-02:
I upgraded my intel Hackintosh from Sonoma to Sequioa and the first two weeks the itlwm and heliport wifi worked, however it suddenly stopped working. So I dont know what realy happened.

BigFoxses, 2024-10-07:
have you tried https://openintelwireless.github.io/General/Troubleshooting.html ? Clear preference settings might help

FocusSK, 2024-10-08:
I'm facing the same issue on the same PC!

emanuele-pogliari, 2024-10-08:
> have you tried https://openintelwireless.github.io/General/Troubleshooting.html ? Clear preference settings might help

i think I did. Nothing worked.
[image]
maybe it could be a usb map problem? 
I think i did all correctly with the mapping....

> I'm facing the same issue on the same PC! My WiFi Card Model: Intel AX200.

My pc? this SER 5?

ethans65, 2025-02-15:
same thing happening here, card is realtek 8821ce.