OpenIntelWireless/itlwm #1017: For Intel AX210, fixing the broken configuration. (airportItlwm.kext & Sonoma 14.5+ )

atlntq, 2024-09-30:
After the Sonoma 14.5 update, many users experienced wifi connection problems. I was one of them. I got everything back in place with “RT2870USBWirelessDriver.kext” among the kext.s I used completely by chance. moreover, AirDrop works smoothly. for your information. if it doesn't solve the problem for you, I can share my own EFI folder, friends. 
pay attention to the order, airportltlwm.kext is down below as usual, RT2870 kext is working with SMCBattery on it. I moved it a few times and everything is broken again.

Stay with love. by the way I apologize for my English.

[image]
[image]
[image]

atlntq, 2024-09-30:
there is a small wording error, “RT2870USBWirelessDriver.kext” will not be processed into plist, it will only be found in the Opencore kext folder. it's ridiculous but it works :)

jzrodriguez98, 2024-09-30:
From where you got this kext (RT2870USBWirelessDriver.kext)? Why are you using it? What Intel wifi card do you have in your hackintosh?

atlntq, 2024-10-01:
I am sharing my complete EFI folder. I think friends with more information can clarify the issue. 
Correction: the wifi card I use is “Intel AX201”

[https://www.dropbox.com/t/htMgw6vtCCCfVyhI](url)

Lekki369, 2024-10-02:
Hi @atlntq on sonoma can you tell me what's not working on your build

viniciusmi00, 2024-10-10:
> I am sharing my complete EFI folder. I think friends with more information can clarify the issue. Correction: the wifi card I use is “Intel AX201”
> 
> [https://www.dropbox.com/t/htMgw6vtCCCfVyhI](url)

Can you re-upload?
Have you tested it on Sequoia?

closed 2025-05-27 by zxystd