OpenIntelWireless/itlwm #1020: Heliport Crash on Sequoia 15.0

Serverbd-Technology, 2024-11-03:
**Environment**

- Kext Version:  2.0.0
- WiFi Card Model: Intel AX200
- PCI Product ID: 0X24FD
- macOS Version: 15.0 Beta (24A5327a)

**Description**
In my HP G6 laptop, Intel WiFi was working perfectly. Last night, the laptop unexpectedly restarted due to a battery issue, and now the WiFi won’t start. I removed HeliPort using CCleaner’s uninstall option and reinstalled it, but still no luck.

It happened 2/3 time before ReinstallOS is fixed past. One of my friend and one small brother also face same issue with Sequoia Heliport they have different model of HP laptop.

**Kext Download Source**
https://github.com/OpenIntelWireless/HeliPort/releases/download/v1.5.0/HeliPort.dmg
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.0.0/itlwm_v2.0.0_stable.kext.zip

mysongbd, 2024-11-03:
Same issue with me. kindly fixed it.

h0smed, 2024-11-11:
Same issue here

zxystd, 2024-11-26:
can you upload the running log? follow the troubleshot instructions here.
https://openintelwireless.github.io/itlwm/Troubleshooting.html

kevinchenxin, 2024-12-16:
bluetooth work?

mysongbd, 2024-12-16:
All work