OpenIntelWireless/itlwm #1022: 802.11ac Wireless-AC 9560 can only connect to wifi and cannot turn on Bluetooth

hanmafalali, 2024-11-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
macOS
- Kext Version: AirportItlwm.kext v2.3.0（Ventura）
- WiFi Card Model: 802.11ac Wireless-AC 9560
- PCI Product ID: 
- macOS Version: macOS Ventura 13.4

**Description**
My inter wireless network card module is built into the motherboard, the motherboard model is TUF B360-PRO GAMING (WI-FI)，After I configured AirportItlwm.kext v2.3.0 (Ventura), I restarted my computer and it showed that there was WIFI and it could be used normally. However, the Bluetooth switch in the upper right corner showed that it was turned on, but I found it was turned off in the West Yong settings. When I returned to the desktop to check the Bluetooth switch in the upper right corner, I found that it was automatically turned off. How can I adjust it? Thank you! You have worked hard~

**Bug Report Archive**

**Kext Download Source**

nsby494, 2024-11-19:
wifi驱动和蓝牙驱动是不同的，airportitlwm.kext并不包括蓝牙驱动。kenerl列表中加入三个蓝牙驱动：BlueToolFixup.kext、IntelBTPatcher.kext、IntelBluetoothFirmware.kext，注意USB定制情况，如果三个驱动加上还没有蓝牙，建议用通用USB驱动USBInjectAll.kext

zxystd, 2024-11-26:
you should use bluetooth firmware drivers in additionally.

closed 2024-11-26 by zxystd