OpenIntelWireless/itlwm #1028: Pls add support to macOS Sequoia and Sequoia Beta for airportitwlm

uptownzombi21, 2025-01-09:
the thing is i gotta use airdrop, iPhone mirroring, and iMessage. Please develop a version for Sequoia as fast as possible

perez987, 2025-01-09:
Duplicated of

#1009 

Go here to follow the topic.

ameenjuz, 2025-01-10:
please add support for macOS Sequoia

uptownzombi21, 2025-01-10:
ik i have tried heliport + itwml but I cant use airdrop

also my PC used to run ventura and sonoma duel boot and they can all use airdrop bc of airportitwlm,  but there ain't airportitwlm in sequoia

uptownzombi21, 2025-01-10:
also i found u on insanelymac >:D @perez987

perez987, 2025-01-10:
@uptownzombi21 
That is my GitHub but it can't help you with your request.

For many users, the lack of AirportItwlm support on Sequoia is an inconvenience. That's why the issue I referred to has 124 comments, because of the interest in this issue.
I refer you to that issue because that's where you can find more information.

You probably already know this but there is a way to use AirportItlwm-Ventura on Sonoma that is described in one of the comments on the other issue, see [here](https://github.com/OpenIntelWireless/itlwm/issues/1009#issuecomment-2370919270).

uptownzombi21, 2025-01-11:
im not using dortania ;-;

uptownzombi21, 2025-01-12:
also checkout my website. theres ab surprise in the setting link from the dropdown

perez987, 2025-01-16:
@uptownzombi21 
Please close this issue. We'll follow the topic in the other issue.

closed 2025-01-21 by uptownzombi21