OpenIntelWireless/itlwm #1029: Please add AirportItlwm support for macOS 15 Sequoia +1

xxlarc33, 2025-01-10:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 
- WiFi Card Model: 
- PCI Product ID: 
- macOS Version: 

**Description**

**Bug Report Archive**

**Kext Download Source**

wirl, 2025-01-13:
It's dead Jim. This repo hasn't seen any changes in 7 months. If you guys are lucky someone knowledgeable forks this and continues development.

xxlarc33, 2025-01-27:
when die?

xxlarc33, 2025-01-27:
how old was he?

uptownzombi21, 2025-01-27:
lol @xxlarc33 

someone actually already tried to continue the development, but they all failed

here's a link to the most recent "attempt" to recreate the[ intel WIFI environment](https://github.com/Lorys89/itlwm)

idk if it works, never tried it, but I hope u guys can give it a try( I'm on a hackintosh and windows duel boot, I accidentally erased my macOS hard drive to test my custom[ AtlasOS](https://github.com/Atlas-OS/Atlas) playbook, and I'm kinda too lazy to install macOS again, so I hope u guys can help me test it :DD)

Brad1944, 2025-03-11:
Well if it were that easy to update it, it won't take anyone more than a month to atleast make it "work".

Apple had changed lots of things on IO80211Family by the release of sequoia. You are free to contribute, but I'm sure that you need atleast a somewhat good knowledge on reverse engineering to do even a "helpful" progress at this

zhangcheng142, 2026-10-02:
[https://github.com/zhangcheng142/AirportItlwm-Sequoia](https://github.com/OpenIntelWireless/itlwm/issues/AirportItlwm-Sequoia) worked on macOS Sequoia 15.8