OpenIntelWireless/itlwm #1030: Add or provide a way to implement Broadcom wifi cards

lost-a-byte, 2025-01-11:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: None
- WiFi Card Model: **Broadcom 943602**
- PCI Product ID: 
- macOS Version: Sequoia

**Description**
Hello itlwm developers!  

Can we add some obsolete Broadcom wifi cards into itlwm kext to use Internet from Recovery
Or someone give me a hint to do that?

Since Apple removed support for a lot of Broadcom wifi cards in the latest macOS, I think that we have to find a way to use Broadcom Wifi card in Recovery mode, or even use the card without disabling SIP and installing patches.

I see OpenBSD does have support for Broadcom 943602 
**Bug Report Archive**
None

**Kext Download Source**
None

Malumen, 2025-01-14:
This might almost constitute an enditely different `kext`, no? 

Also, project is dead.

zxystd, 2025-01-14:
I have done this last year, but not open source, because only full-mac devices supported. maybe open source if I am available.

lost-a-byte, 2025-03-15:
Yes, there are a lot of full-mac devices that openbsd supporting
| [Chipset](https://man.openbsd.org/bwfm#Chipset) | [Spectrum](https://man.openbsd.org/bwfm#Spectrum) |  [Type](https://man.openbsd.org/bwfm#Type) | [MIMO](https://man.openbsd.org/bwfm#MIMO) | [Bus](https://man.openbsd.org/bwfm#Bus) | 
| -------- | ------- | ------- | ------- | ------- |
| BCM43143 | 2GHz | 11n | 	1x1 |	SDMMC/USB |
| BCM43236| 2GHz/5GHz | 11n | 	2x2 |	USB |
| BCM4324 | 2GHz/5GHz | 11n | 	2x2 |	SDMMC |
| BCM43242| 2GHz/5GHz | 11n | 	2x2 |	USB |
| BCM4329 | 2GHz/5GHz | 11n | 	2x2 |	SDMMC |
| BCM4330 | 2GHz/5GHz | 11n | 	2x2 |	SDMMC |
| BCM4334 | 2GHz/5GHz | 11n | 	2x2 |	SDMMC |
| BCM43340 | 2GHz/5GHz | 11n | 	1x1 |	SDMMC |
| BCM43341| 2GHz/5GHz | 11n | 	1x1 |	SDMMC |
| BCM4335 | 2GHz/5GHz | 11ac | 	1x1 |	SDMMC |
| BCM43362 | 2GHz | 11n | 	1x1 |	SDMMC |
| BCM43364| 2GHz | 11n | 	1x1 |	SDMMC |
| BCM4339 | 2GHz/5GHz | 11ac | 	1x1 |	SDMMC |
| BCM43430 | 2GHz | 11n | 	1x1 |	SDMMC |
| BCM43455 | 2GHz/5GHz | 11ac | 	1x1 |	SDMMC |
| BCM43456| 2GHz/5GHz | 11ac | 	2x2 |	SDMMC |
| BCM4350 | 2GHz/5GHz | 11ac | 	2x2 |	PCI |
| BCM4354 | 2GHz/5GHz | 11ac | 	2x2 |	SDMMC |
| BCM4356 | 2GHz/5GHz | 11ac | 	2x2 |	PCI/SDMMC |
| BCM43569 | 2GHz/5GHz | 11ac | 	2x2 |	USB |
| BCM43602| 2GHz/5GHz | 11ac | 	3x3 |	PCI |
| BCM4371 | 2GHz/5GHz | 11ac | 	2x2 |	PCI |
| BCM4378 | 2GHz/5GHz | 11ax | 	2x2 |	PCI |
| BCM4387 |	2GHz/5GHz | 11ax | 	2x2 |	PCI |

I think that we can update the build target to build Broadcom device alongside with Intel device to reduce bundle size (if possible), eg: itlwm.kext and bcmwm.kext