OpenIntelWireless/itlwm #1031: Intel(R) Wi-Fi 6 AX210 160MHz not work in Sonoma 14.7.4

xyxu, 2025-02-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip
- WiFi Card Model: Intel(R) Wi-Fi 6 AX210 160MHz
- PCI Product ID: 2725/0024
- macOS Version: 14.7.4

**Description**
no network interface created with Intel(R) Wi-Fi 6 AX210 160MHz in Sonoma 14.7.4

**Bug Report Archive**

[itlwm.log](https://github.com/user-attachments/files/18766669/itlwm.log)

**Kext Download Source**

zxystd, 2025-02-12:
`[  104.337324]: itlwm: : acquiring device failed`
PCI device not correct. check your EFI or PCI/ACPI related settings.

mishealgo, 2025-03-10:
probably  this result can help on continuo on the project for macOS Sequoia Support  : 

 I have tried this link and I rebuild the file but still only works on MacOS Sonoma. I tried on MacOS Sequoia after I changed the ownership file to root to try to get it loaded but it failed and got this error :

Code=31 "Error occurred while building a collection:

1: One or more binaries have an error that prevents linking. See other errors.
2: Could not use 'com.zxystd.AirportItlwm' because: Failed to bind '_ZN17IO80211Controller11postMessageEjPvmjS0' in 'com.zxystd.AirportItlwm' (at offset 0x2568 in __DATA, __const) as could not find a kext which exports this symbol
com.zxystd.AirportItlwm specific:
1: Failed to bind '_ZN17IO80211Controller11postMessageEjPvmjS0' in 'com.zxystd.AirportItlwm' (at offset 0x2568 in __DATA, __const) as could not find a kext which exports this symbol
" UserInfo={NSLocalizedDescription=Error occurred while building a collection:
1: One or more binaries has an error which prevented linking. See other errors.
2: Could not use 'com.zxystd.AirportItlwm' because: Failed to bind '_ZN17IO80211Controller11postMessageEjPvmjS0' in 'com.zxystd.AirportItlwm' (at offset 0x2568 in __DATA, __const) as could not find a kext which exports this symbol
com.zxystd.AirportItlwm specific:
1: Failed to bind '_ZN17IO80211Controller11postMessageEjPvmjS0' in 'com.zxystd.AirportItlwm' (at offset 0x2568 in __DATA, __const) as could not find a kext which exports this symbol.

*### *after using ChatGPT it tells me about that error:**

Missing IO80211Controller::postMessage in macOS Sequoia
The error means that AirportItlwm.kext is trying to use a function (postMessage) that no longer exists in IO80211Controller (part of IO80211Family.kext). This suggests that Apple changed or removed the function in macOS Sequoia.

Possible Locations Where postMessage Moved:

AppleBCMWLANCore.kext (For Broadcom-based Wi-Fi)
This kext contains Apple’s internal Wi-Fi handling for Broadcom chipsets.
Check if it exists:
sudo find /System/Library/Extensions -name "AppleBCMWLANCore.kext"

DriverKit-based Wi-Fi Drivers
Apple may have migrated Wi-Fi handling from IO80211Family.kext to DriverKit extensions.
Check the installed DriverKit extensions:
systemextensionsctl list
Look for any com.apple.driverkit entries related to Wi-Fi.

AppleIntelWiFi.kext (If Intel Support Exists)
Apple might have started using a dedicated Intel Wi-Fi driver.
Check for it:
sudo find /System/Library/Extensions -name "AppleIntelWiFi.kext"

Next Steps to Fix AirportItlwm.kext

Reverse Engineer Function Changes
If IO80211Controller::postMessage is missing, find the new function that replaced it.
Dump symbols from IO80211Family.kext in Sequoia:
nm -gU /System/Library/Extensions/IO80211Family.kext/Contents/MacOS/IO80211Family | grep postMessage
If no results, the function was removed or renamed.

Patch AirportItlwm.kext
If postMessage was moved or renamed, modify AirportItlwm.kext to use the new function.
If it was removed, a workaround is needed (e.g., patching AirportItlwm to avoid calling it).

junuo-S, 2025-05-20:
May I ask if this issue has been resolved? I also encountered the same problem in macOS 14.7.4 and don't know how to solve it

closed 2025-05-27 by zxystd