OpenIntelWireless/itlwm #1033: slow wifi in Sonoma 14.7

SAHILDUDHAL21, 2025-04-14:
internet speeds are at most 200kbs 
I tried itlwm and airportitlwm

mdtamim54133, 2025-04-15:
> internet speeds are at most 200kbs I tried itlwm and airportitlwm

will airportitlwm work on macos 14.7.5?

arjunlatiyal, 2025-04-18:
> internet speeds are at most 200kbs I tried itlwm and airportitlwm

मोबाइल हॉटस्पॉट(एयरटेल) से स्पीड स्लो ही आती है मेरे भी, लेकिन फ़ाइबर वाईफ़ाई से फुल स्पीड आती है मेरे ।

SAHILDUDHAL21, 2025-04-18:
> > internet speeds are at most 200kbs I tried itlwm and airportitlwm
> 
> will airportitlwm work on macos 14.7.5?

It works but speed is low sometimes it gets max 10 MBps

SAHILDUDHAL21, 2025-04-18:
> > internet speeds are at most 200kbs I tried itlwm and airportitlwm
> 
> मोबाइल हॉटस्पॉट(एयरटेल) से स्पीड स्लो ही आती है मेरे भी, लेकिन फ़ाइबर वाईफ़ाई से फुल स्पीड आती है मेरे ।

I tried airtel jio and wifi 
Airtel is lowest 
Jio 1 mbps
Wifi router max 5 to 10 MBps

closed 2025-04-30 by SAHILDUDHAL21