OpenIntelWireless/itlwm #1034: Please add AirportItlwm support for macOS 15 Sequoia 15.4

mishealgo, 2025-04-27:
Please If you have found a way to get Airportitlwn.kext on Macos sequoia share it with us.
I can help here for testing if you need to or put us on the way or mention the tools so we can help out .
or Let us know if this project is dead or what

Blu-Dread, 2025-04-29:
i am trying to build it using this guide on YouTube

[](https://www.youtube.com/watch?v=_CaCxPV1GHo)

but it's difficult because it's a very old video

jzrodriguez98, 2025-04-29:
> i am trying to build it using this guide on YouTube
> 
> 
> [](https://www.youtube.com/watch?v=_CaCxPV1GHo)
> 
> https://www.youtube.com/watch?v=_CaCxPV1GHo but it's difficult because it's a very old video
> 

@Blu-Dread Have you checked the very first issue that was created requesting the developer to work on making Airportitlwm kext to work on macOS Sequoia? [](https://github.com/OpenIntelWireless/itlwm/issues/983) here various hackintosh enthusiasts and devs talked about the changes made by Apple which caused the kext to not work in Sequoia.

mishealgo, 2025-04-29:
> > i am trying to build it using this guide on YouTube
> > 
> > https://www.youtube.com/watch?v=_CaCxPV1GHo, but it's difficult because it's a very old video
> 
> [@Blu-Dread](https://github.com/Blu-Dread) Have you checked the very first issue that was created requesting the developer to work on making Airportitlwm kext to work on macOS Sequoia? [#983](https://github.com/OpenIntelWireless/itlwm/issues/983) Here, various hackintosh enthusiasts and devs talked about the changes made by Apple, which caused the kext to not work in Sequoia. If you haven't looked at it, maybe the discussion in issue [#983](https://github.com/OpenIntelWireless/itlwm/issues/983) will represent additional context that is relevant to work on a kext version that works in Sequoia.

Hi jzrodriguez98

Thank you for your reply, but I have seen no update on it since October 2024, and [#983] most of their comments are talking about itilwm + hillop what whatever the name, and they forgot to focus on our main goal, which is getting the Airportitlwm project to continue again ( the native wireless  connection way ), that way I had to create this threat to give attention to that project and alot of us here ready to help

jzrodriguez98, 2025-04-29:
@Blu-Dread It's great that you are trying to make the Airportitlwm kext working in Sequoia. As @mishealgo said, the main issue is what Apple did, using DriverKit SDK to build drivers for wifi, which is what some of the devs and other users mentioned in issue 983. You're right, since then users have been focused on either using OCLP to use in Sequoia the old Airportitlwm kext created in Ventura or setting up itlwm+heliport for use in Sequoia.

zxystd, 2025-05-20:
Apple not leave the interface that communicates to user space in IO80211Family framework, their programers hardcode the implementation class for Broadcom driver. tired to hack their shit(maybe use Lilu).

Brad1944, 2025-05-20:
> Apple not leave the interface that communicates to user space in IO80211Family framework, their programers hardcode the implementation class for Broadcom driver. tired to hack their shit(maybe use Lilu).

So for sequoia we need to completely recode the whole codebase? to not use IO80211 and use Lilu instead?

zxystd, 2025-05-20:
> So for sequoia we need to completely recode the whole codebase? to not use IO80211 and use Lilu instead?

no, just the IOCTL section, for now we use virtual table to implement the actual communication, see here https://github.com/OpenIntelWireless/itlwm/blob/master/AirportItlwm/AirportItlwmSkywalkInterface.hpp

but Apple not provide thirdparty driver to link the methods at all, so we have no way to communicate with user space application.s There may have a chance to use Lilu to hook every ioctl methods and replace to our drivers code.

Daasin, 2025-06-18:
> but Apple not provide thirdparty driver to link the methods at all, so we have no way to communicate with user space application.s There may have a chance to use Lilu to hook every ioctl methods and [...]

@zxystd Would it be worth skipping over sequoia entirely now that Tahoe is said to be the last Intel supported macOS? It'd make sense for that last hackintoshable version to be the main focus, right? (Although I know the whole system can change with it being early dev build/previews)

zhangcheng142, 2026-10-02:
[https://github.com/zhangcheng142/AirportItlwm-Sequoia](AirportItlwm-Sequoia) worked on macOS Sequoia 15.8