OpenIntelWireless/itlwm #1037: Please add AirportItlwm support for macOS 16

Xrandon, 2025-06-02:
MacOS 16 is being announced soon. Please start working on adding support, otherwise I wont be able to use macOS 16 with wifi immediately after it releases.

parasaryan, 2025-06-03:
Entitled Clown

zxystd, 2025-06-04:
Do you know what you are talking about?

closed 2025-06-04 by zxystd

Xrandon, 2025-06-10:
okay this thread needs to be reopened because macOS 26 (Tohoe) just got released. @zxystd

jzrodriguez98, 2025-06-10:
> okay this thread needs to be reopened because macOS 26 (Tohoe) just got released. @zxystd 

We even don't have Airportitlwm support in Sequoia given the high level of changes made by Apple moving everything to the userspace. The developer is still working on Sequoia support, so let's be patient and don't start asking for macOS 26 support. Lilu kext has not been fully updated to support macOS 26 anyway.

5T33Z0, 2025-06-11:
So, I managed to install macOS Tahoe and tested the following in macOS 26:

- Blocking `com.apple.iokit.IOSkywalkFamily` seems to work – the system doesn't crash
- Injecting `IOSkywalkFamily.kext` and `IO80211FamilyLegacy.kext` works as well
- Then I installed the latest Kernel Develpment Kit and OCLP 2.4.0 nightly
- Then I wanred to test Modern Wi-Fi patching but I am held back by OCLP: <br>[image]